intel/zephyr.js #1218: [net] Can not distinguish IPv4 address precisely

cuiyanx, 2017-06-07:
### Description

When set IPv4 address as "192.168.0.2017", I have got 'true' using `net.isIPv4` function and got '4' using `net.isIP` function. But the IPv4 address space is 0~255. I think we would better to distinguish IPv4 address precisely.
 

### Test Code

Simple test:
```
var net = require("net");

var IPv4Address = "192.168.0.2017";
var IPResult1;

IPResult1 = net.isIP(IPv4Address);
console.log("'192.168.0.2017' ---- type: " + typeof IPResult1 + " value: " + IPResult1);

IPResult1 = net.isIPv4(IPv4Address);
console.log("'192.168.0.2017' ---- type: " + typeof IPResult1 + " value: " + IPResult1);
```

 

### Steps to Reproduction

 

### Actual Result

[image]

 

### Expected Result

Can distinguish IPv4 address precisely
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | fee4637 | Arduino 101 | June 6, 2017 | Fail

 

### Additional Information

jprestwo, 2017-06-07:
This looks to be a zephyr bug. The function we are using to verify the address (`net_addr_pton`) does the following:
```
len = strlen(src);
for (i = 0; i < len; i++) {
    if (!(src[i] >= '0' && src[i] <= '9') &&
	src[i] != '.') {
	    return -EINVAL;
    }
}
```

So its only checking that the string only consists of numbers and periods. So any string that only contains numbers and periods would succeed.

cuiyanx, 2017-06-08:
@jprestwo Can we fix it?