intel/zephyr.js #1219: Can't build HCI compatible BLE firmware from the included deps/zephyr folder

ttzeng, 2017-06-07:
I'm told need to [flash the HCI compatible BLE firmware](https://www.zephyrproject.org/doc/boards/x86/arduino_101/doc/board.html#flashing-the-bluetooth-core) in order to use the Zephyr BLE function, but trying to build the firmware using the included zephyr repo in ZJS got the following errors:
```
make[1]: Entering directory '/home/tzeng/Workspace/zephyr.js/deps/zephyr'
make[2]: Entering directory '/home/tzeng/Workspace/zephyr.js/deps/zephyr/samples/bluetooth/hci_uart/outdir/arduino_101_ble'
  GEN     ./Makefile
scripts/kconfig/conf --silentoldconfig Kconfig
*
* Restart config...
*
*
* ARM Cortex-M options
*
Align stacks on double-words (8 octets) (STACK_ALIGN_DOUBLE_WORD) [Y/n/?] y
Attach an NMI handler at runtime (RUNTIME_NMI) [N/y/?] n
Fault dump level (FAULT_DUMP) [2] 2
SRAM Size in kB (SRAM_SIZE) [16] 16
SRAM Base Address (SRAM_BASE_ADDRESS) [0x20000000] 0x20000000
Flash Size in kB (FLASH_SIZE) [256] 256
Flash Base Address (FLASH_BASE_ADDRESS) [0x00000000] 0x00000000
Kernel load offset (FLASH_LOAD_OFFSET) [0] (NEW) 
```

jprestwo, 2017-06-07:
Try removing the `.config` file in that sample directory.

ttzeng, 2017-06-08:
@jprestwo thanks, it works.

closed 2017-06-08 by ttzeng

grgustaf, 2017-06-08:
Yeah, seems to work for me. Try 'make pristine' in that directory? I run `make BOARD=arduino_101_ble` and it works.