intel/zephyr.js #1226: [ashell] No error thrown when require a not existing module in ashell

qiaojingx, 2017-06-09:
### Description

Error is not thrown when require a module which not stored in board.
### Test Code

[ashell.md](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md).
 
### Steps to Reproduction

1. `make ide`
2. `make dfu`
3. connect A101 with https://01org.github.io/zephyrjs-ide/#/editor
4. Run code `var temp = require('Temp.js');` (make sure Temp.js is not stored in board)
 
### Actual Result

No error thrown.
[image]
 
### Expected Result

Module not found error should be thrown. 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | fee4637 | Arduino 101 | Jun 9, 2017 | Fail
 
### Additional Information

If require a stored module which is not standards compliant, it will thrown `NotSupportedError: module not found 'xxx'` . This error message will confuse user.
[image]

brianjjones, 2017-06-09:
Shouldn't be too hard to add.

brianjjones, 2017-07-05:
Sorry for the slow reply, got caught up in other fixes.  This should take care of it. https://github.com/01org/zephyr.js/pull/1330

qiaojingx, 2017-07-17:
Verified with commit af7eadd. This issue is not reproduced.

closed 2017-07-17 by qiaojingx