intel/zephyr.js #1229: Zephyr SDK 0.9.1 does not work

poussa, 2017-06-09:
Using Zephyr SDK 0.9.1 produces unbootable image. Nothing on console. Tried with make ide ROM=256 and make JS=samples/HelloWorld.js

uhofemei, 2017-06-09:
I just used SDK 0.9.1 to build Smart Home PR #150 and I was able to boot.

jimmy-huang, 2017-06-09:
@poussa see comment on #1204, anything with AIO is broken due to the Zephyr if you upgrade to 9.1, one way to get around this is adding debug to arc image, so adding CONFIG_DEBUG=y to arc/fragments/prj.conf.base

uhofemei, 2017-06-09:
This is not set for the PR 01org/SmartHome-Demo#150 I mentioned above and it still works.

grgustaf, 2017-06-14:
As with the 0.9.0 foibles, we'll just stick with recommending 0.9.0 until there's a decent resolution to this in Zephyr.

qiaojingx, 2017-07-10:
AIO related samples are broken due to the Zephyr issue(https://jira.zephyrproject.org/browse/ZEP-2363) after commit 0334065(zephyr update to 1.8 and zephyrSDK update to 0.9.1). This issue blocks ashell and ide testing.
Platform: Arduino 101
Commit: 05fda87

qiaojingx, 2017-07-13:
Verified with commit 1ff882b + #1355 . This issue is not reproducible.

commit 5f390051 (2017-07-14): [build] Work-around fix for Zephyr SDK bug on AIO (#1355)

The 0.9.1 SDK still has upstream bug #2363 not fixed, so
this work-around patches our build to build with -O2 on ARC
images so it fixes the bug with AIO not booting.

Fixes #1229
closed 2017-07-14 by grgustaf