intel/zephyr.js #1235: BME280 example

kenchris, 2017-06-12:
Now we have BMP280 example, it would be nice to use the new BME280 instead as it has temp/pressure and relative humidity.

https://www.adafruit.com/product/2652
https://www.sparkfun.com/products/13676

Of course having it integrated with generic sensors would be even cooler :-) This is a super exciting sensor!

kenchris, 2017-06-12:
Related https://jira.zephyrproject.org/browse/ZEP-130

kenchris, 2017-06-12:
Actually it is supported directly by Zephyr already: https://jira.zephyrproject.org/browse/ZEP-131

jimmy-huang, 2017-06-12:
@kenchris we'll need to get 1 of these BMP280 sensors first, it's on our hardware list.

grgustaf, 2017-06-14:
@kenchris when you said 'instead' what did you mean? Replace the temp sensor in the WebBT demo?

kenchris, 2017-06-14:
Just ignore the instead - not sure why I added that

grgustaf, 2017-06-14:
With my habit of quick skimming I didn't notice the difference between BMP280 and BME280 - makes sense now. :)