intel/zephyr.js #1237: Accessing multiple sensors unstable

uhofemei, 2017-06-12:
This bug is based on PR https://github.com/01org/SmartHome-Demo/pull/150. 

The BLE connection keeps dropping in <= 3 mins after it has been established via the command line using the instructions here:  https://github.com/01org/zephyr.js/blob/master/docs/6lowpan-ble.md

While the connection is established, accessing more than 1 sensor in parallel via the REST API service will cause the micro controller to lock up.

jprestwo, 2017-06-14:
The following config options definitely helped with this. It allowed you to access multiple at once, but if you continue to stress the system (many multi resource accesses) it will eventually lock up again:

```
CONFIG_BLUETOOTH_L2CAP_TX_BUF_COUNT=20
CONFIG_BLUETOOTH_RX_BUF_COUNT=20
```

jimmy-huang, 2017-07-28:
@uhofemei the patch #1417 may fixed this, please retest

gvancuts, 2017-08-02:
Hi @jimmy-huang @uhofemei , has it been determined if #1417 fixes this?

jimmy-huang, 2017-08-02:
@gvancuts It's pretty stable for us running OcfServer and OcfSensorServer, but I don't have any setup of the smarthome demo, so it is best for @uhofemei to test it and report back.

uhofemei, 2017-08-02:
Will do.

uhofemei, 2017-08-15:
Accessing an Arduino 101 with ZJS and multi server configuration via the cloud application will bring down the Zephyr device within 5 seconds.

uhofemei, 2017-08-15:
The console output doesn't reveal why it crashed.

`ipm_console0: 'Sensor core running ZJS ARC support image'

Started OCF server
Light resource registered
Temperature resource registered
Connected
oc_network_receive: received 26 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
oc_send_buffer: outgoing message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
oc_send_buffer: failed: (119)
oc_network_receive: received 26 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
oc_send_buffer: outgoing message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
oc_send_buffer: failed: (119)
oc_network_receive: received 26 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
oc_send_buffer: outgoing message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
oc_send_buffer: failed: (119)
oc_network_receive: received 26 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
oc_send_buffer: outgoing message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
oc_send_buffer: failed: (119)
oc_network_receive: received 25 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
oc_network_receive: received 24 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
oc_send_buffer: outgoing message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
oc_send_buffer: sent 87 bytes
oc_network_receive: received 61 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:45798
oc_network_receive: received 61 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:45447
oc_network_receive: received 61 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
oc_network_receive: received 61 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:45798
oc_network_receive: received 61 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:45447
oc_network_receive: received 61 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
oc_network_receive: received 61 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:45798
oc_network_receive: received 61 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:45447
oc_network_receive: received 61 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:45798
oc_network_receive: received 61 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:45447
oc_network_receive: received 61 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
oc_network_receive: received 26 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
oc_send_buffer: outgoing message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
oc_send_buffer: failed: (119)
oc_network_receive: received 26 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
oc_send_buffer: outgoing message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
oc_send_buffer: failed: (119)
oc_network_receive: received 72 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
ocf_get_handler()
oc_send_buffer: outgoing message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
oc_send_buffer: sent 25 bytes
oc_network_receive: received 73 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
ocf_get_handler()
oc_send_buffer: outgoing message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
oc_send_buffer: sent 26 bytes
[... 33 lines cut ...]
oc_network_receive: received 73 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
ocf_get_handler()
ocf_get_handler()
oc_send_buffer: outgoing message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
oc_send_buffer: sent 26 bytes
oc_network_receive: received 26 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
ocf_get_handler()
ocf_get_handler()
oc_send_buffer: outgoing message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
oc_send_buffer: failed: (119)
oc_network_receive: received 26 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
ocf_get_handler()
ocf_get_handler()
oc_send_buffer: outgoing message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
oc_send_buffer: failed: (119)
oc_network_receive: received 26 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
ocf_get_handler()
ocf_get_handler()
oc_send_buffer: outgoing message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009
oc_network_receive: received 26 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe7f:d746]:38009`

jimmy-huang, 2018-01-25:
@uhofemei Now with Zephyr 1.10, bluetooth should be much more stable now, please re-test.