intel/zephyr.js #1244: [pme] The number of committed neurons in the network can be got as '0'

cuiyanx, 2017-06-14:
### Description

`PME.getCommittedCount` in [/docs/pme.md](https://github.com/01org/zephyr.js/blob/master/docs/pme.md):
```

PME.getCommittedCount

unsigned short getCommittedCount();

Gets the number of committed neurons in the network.
Returns the number of comitted neurons in the network (a value between 1-128).
```
 But, I can get The number of committed neurons in the network as '0'

### Test Code

Simple test:
```
var pme = require("pme");
console.log("Committed Count: " + pme.getCommittedCount());
```

 

### Steps to Reproduction

 

### Actual Result

```
ipm_console0: 'Sensor core running ZJS ARC support image'
Committed Count: 0
```

### Expected Result

Update pme.md doc.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 819a41a | Arduino 101 | June 14, 2017 | Fail

 

### Additional Information

jimmy-huang, 2017-06-28:
I think this is a typo, should be 0-128, I'll update the API docs

closed 2017-06-29 by grgustaf

cuiyanx, 2018-01-10:
Fixed and closed

PR #1321 (2017-06-29): [doc] Fix typo in PME doc
The number of committed neurons can be 0-128.

Fixes #1244
commit d92cba55 (2017-06-29): [doc] Fix typo in PME doc (#1321)

The number of committed neurons can be 0-128, also
fixed parameter type to be consistent.

Fixes #1244, #1249