intel/zephyr.js #1245: [pme] Can not change neuron context value

cuiyanx, 2017-06-14:
### Description

I can set neuron context value and no error threw out, but I always get neuron context value as '127'.
 

### Test Code

Simple test:

```
var pme = require("pme");

pme.setNeuronContext(1);
console.log("neuron context: " + pme.getNeuronContext());
pme.setNeuronContext(10);
console.log("neuron context: " + pme.getNeuronContext());
pme.setNeuronContext(100);
console.log("neuron context: " + pme.getNeuronContext());
```
 

### Steps to Reproduction

 

### Actual Result

```
ipm_console0: 'Sensor core running ZJS ARC support image'
neuron context: 127
neuron context: 127
neuron context: 127
```

### Expected Result

 Can be changed.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 819a41a | Arduino 101 | June 14, 2017 | Fail

 

### Additional Information

jimmy-huang, 2017-06-28:
I think this is actually an internal API that's used for restoring context, as it only works in save/restore modes, so obviously it's not going to work if you call it directly.   I think we should actually drop the set/get neuron context api, as the regular app just calls saveNeurons() and restoreNeurons(), they don't just set/get the neuron context.

cuiyanx, 2017-07-10:
@jimmy-huang OK, wait for PME doc updated. I will close this issus and remove the test point.

closed 2017-07-24 by grgustaf

cuiyanx, 2017-07-25:
Remove the test point and closed

PR #1376 (2017-07-24): [pme] Drop neuron context APIs as it's internal-only
The setNeuronContext() and getNeuronContext() APIs are internal APIs
that should only be called in save/restore modes, applications
should not be calling this function directly, but instead
use the saveNeurons() and restoreNeurons() instead.

Fixes #1245, #1246
commit b8662364 (2017-07-24): [pme] Drop neuron context APIs as it's internal-only (#1376)

The setNeuronContext() and getNeuronContext() APIs are internal APIs
that should only be called in save/restore modes, applications
should not be calling this function directly, but instead
use the saveNeurons() and restoreNeurons() instead.

Also dropping setGlobalContext() which is internal as well.

Fixes #1245, #1246