intel/zephyr.js #1246: [pme] How can I get neuron context value and global context value as '0'

cuiyanx, 2017-06-14:
### Description

I can set neuron context value and global context value between 1-127, but can get those value between 0-127. I set those value as '0', but throw out error. So, How can I get 0? 

In [/docs/pme.md](https://github.com/01org/zephyr.js/blob/master/docs/pme.md):
```
PME.getGlobalContext

unsigned short getGlobalContext();
Reads the Global Context Register.
Returns the contents of the Global Context Register (a value between 0-127).

PME.setGlobalContext

void setGlobalContext(unsigned short context);
Writes a value to the Global Context Register.
The context is valid context value range between 1-127. A context value of 0 enables all neurons, with no regard to their context.

PME.getNeuronContext

unsigned short getNeuronContext();
Reads the Neuron Context Register.
Returns the contents of the Neuron Context Register (a value between 0-127).

PME.setNeuronContext

void setNeuronContext(unsigned short context);
Writes a value to the Neuron Context Register.
The context is valid context value range between 1-127. A context value of 0 enables all neurons, with no regard to their context.
```

### Test Code

Simple test:
```
var pme = require("pme");

try {
    pme.setGlobalContext(0);
} catch(e) {
    console.log("set global context value as '0'");
}

try {
    pme.setNeuronContext(0);
} catch(e) {
    console.log("set neuron context value as '0'");
}
```

 

### Steps to Reproduction

 

### Actual Result

```
 ipm_console0: 'Sensor core running ZJS ARC support image'

ipm_console0: ''
ipm_console0: 'src/main.c:1106 handle_pme():'
ipm_console0: '(ERROR) context range has to be 1-127'

src/zjs_pme.c:241 zjs_pme_set_global_context():
(ERROR) error code: 2
set global context value as '0'
ipm_console0: ''
ipm_console0: 'src/main.c:1121 handle_pme():'
ipm_console0: '(ERROR) context range has to be 1-127'

src/zjs_pme.c:262 zjs_pme_set_neuron_context():
(ERROR) error code: 2
set neuron context value as '0'
```

### Expected Result

 Can set those value as '0'. 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 819a41a | Arduino 101 | June 14, 2017 | Fail

 

### Additional Information

jimmy-huang, 2017-06-28:
See my comments on #1245, I think we should drop global/neuron APIs as they should be internal.

cuiyanx, 2017-07-10:
@jimmy-huang wait for PME doc updated. I will close this issus and remove the test point.

cuiyanx, 2017-07-25:
Remove the test point and closed

closed 2017-07-25 by cuiyanx