intel/zephyr.js #1249: [pme] There are some error about function parameter description

cuiyanx, 2017-06-15:
### Description

Parameter definition as `number ` in [/docs/pme.md](https://github.com/01org/zephyr.js/blob/master/docs/pme.md):
```
    learn(number[] pattern, unsigned long category);
    unsigned long classify(number[] pattern);
```

Parameter definition as `char ` in [/docs/pme.md](https://github.com/01org/zephyr.js/blob/master/docs/pme.md):
```
 PME.learn

void learn(unsigned char[] pattern, unsigned long category);

Takes a pattern and commits it to the network as training data for a given category.
The pattern is an array of bytes, up to 128 bytes in length. The category indicates to the PME which category this training vector belongs to, that is, if a future input has a sufficiently similar pattern, it will be classified as the same category passed with this pattern.

PME.classify

unsigned long classify(unsigned char[] pattern);

Takes a pattern and uses the committed neurons in the network to classify the pattern.
The pattern is an array of bytes, up to 128 bytes in length.
Returns PME.NO_MATCH if the input data did not match any of the trained categories. Otherwise, the trained category assigned by the network will be returned.
```

### Test Code

 

### Steps to Reproduction

 

### Actual Result

 

### Expected Result

 Definitions and descriptions are consistent

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 819a41a | Arduino 101 | June 15, 2017 | Fail

 

### Additional Information

qiaojingx, 2017-07-04:
Verified with commit 418c871. This issue has been fixed. Parameter definition are `number` in both [Web IDL ](https://github.com/01org/zephyr.js/blob/master/docs/pme.md#web-idl)and [API Documentation ](https://github.com/01org/zephyr.js/blob/master/docs/pme.md#api-documentation)sections.

cuiyanx, 2017-07-10:
fixed and closed

closed 2017-07-10 by cuiyanx