intel/zephyr.js #1255: Failed to find ocf resources after update zephyr.js

qiuzhong, 2017-06-16:
## Description:
After I update the [zephyr.js](https://github.com/01org/zephyr.js) to [44a5e6b](https://github.com/01org/zephyr.js/commit/44a5e6b8547af30583cc324d1263464e4fd76325), all the [SmartHome-Demo zjs-servers](https://github.com/01org/SmartHome-Demo/tree/master/ocf-servers/zjs-servers) samples are not working any more.

## Steps to Reproduce
1. Update the  [zephyr.js](https://github.com/01org/zephyr.js) to [44a5e6b](https://github.com/01org/zephyr.js/commit/44a5e6b8547af30583cc324d1263464e4fd76325), and update its submodules.
2. Build the [led.js samples](https://github.com/01org/SmartHome-Demo/tree/master/ocf-servers/zjs-servers) and flash it.
3. Follow [this](https://github.com/01org/SmartHome-Demo/tree/master/ocf-servers/zjs-servers#establish-a-ble-connection-from-gateway)
4. Install and launch [iot-rest-api-server](https://github.com/01org/iot-rest-api-server), then run `$ cd iot-rest-api-server/test && ./oic-get /res`

## Actual Result:
No LED resource found.

## Expected Result:
The LED resource can be discovered via REST API request.

## Test Builds
| Branch | Commit Id | Target Device | Test Date | Result |
| --- | --- | --- | --- | --- |
| master |	[6180fec](https://github.com/01org/zephyr.js/commit/6180fecad242fa9e097aec207d699d470902b715) | Arduino 101 | June 9, 2017 |	Pass |
| master |	[44a5e6b](https://github.com/01org/zephyr.js/commit/44a5e6b8547af30583cc324d1263464e4fd76325) | Arduino 101 | June 16, 2017 |	Failed |

ttzeng, 2017-06-16:
This can be reproduced by running `samples/OcfServer.js` : the following kernel abort message observed in the serial console:
```
***** Kernel OOPS! *****
Current thread ID = 0xa800fae0
Faulting segment:address = 0x0008:0x400350f9
eax: 0x00000000, ebx: 0xa800daa0, ecx: 0x00000000, edx: 0x00000000
esi: 0x00000000, edi: 0xa800f980, ebp: 0xa800fbec, esp: 0xa800fbdc              
eflags: 0x246                                                                   
Fatal fault in essential thread! Spinning...
```

uhofemei, 2017-06-19:
Is this a ZJS specific issue or would it affect the native multi sensor implementation submitted in https://github.com/01org/SmartHome-Demo/pull/129 as well?

jimmy-huang, 2017-07-06:
I can't reproduce this bug with master:

localhost:8000 /res
[{"di":"b2127280-c8e0-4dcd-91a9-dde617d00ffa","links":[{"href":"/oic/sec/doxm","rt":"oic.r.doxm","if":"oic.if.baseline","p":{"bm":1,"secure":false}}]},{"di":"b2127280-c8e0-4dcd-91a9-dde617d00ffa","links":[{"href":"/oic/sec/pstat","rt":"oic.r.pstat","if":"oic.if.baseline","p":{"bm":1,"secure":false}}]},{"di":"b2127280-c8e0-4dcd-91a9-dde617d00ffa","links":[{"href":"/oic/d","rt":"oic.wk.d","if":"oic.if.baseline","p":{"bm":1,"secure":false}}]},{"di":"b2127280-c8e0-4dcd-91a9-dde617d00ffa","links":[{"href":"/oic/p","rt":"oic.wk.p","if":"oic.if.baseline","p":{"bm":1,"secure":false}}]}]
HTTP: 200

qiuzhong, 2017-07-07:
Actually you did reproduce the issue as there was no LED resource found from you log. Normally the resource list should contain a LED resource like this:
> {"di":"e9df8509-3ee0-4509-4fe0-8509e0e08509","links":[{"href":"/a/led","rt":"/a/led","if":"oic.if.baseline","p":{"bm":1,"secure":false}}]}

The 4 resources in your log are not always present right now.

jimmy-huang, 2017-07-12:
Ok, I thought it was the segfault, but was this working at one point? I checked out commit and I still not see a LED resource

wanghongjuan, 2017-07-12:
It works fine with commit [8d1b343](https://github.com/01org/zephyr.js/commit/8d1b343079d2a03700ab623281acb8b6ea5e1541)

wanghongjuan, 2017-07-20:
@jimmy-huang  Thanks for your clarification

closed 2017-07-28 by grgustaf

qiuzhong, 2017-08-01:
Tried with [e6d1bf8](https://github.com/01org/zephyr.js/commit/e6d1bf82aa4eb7bb12444d5a5f9a4a115ac63c7a), it can work well when there is only one sensor. But with the multi-sensor case, the OCF resources are still not found via REST API.

uhofemei, 2017-08-02:
Looks like that would be a new bug then? Being able to drive more than one sensor at a time is important.

qiuzhong, 2017-08-03:
They're not the same issue. #1237 says the BLE connection is not stable. In my test the BLE connection was quite stable but the OCF resources cannot be found via REST API when there are multiple sensors.
See #1428

PR #1417 (2017-07-28): [ocf] Improved OCF stability over 6lowpan
The OCF was not able to send packets over BLE 6lowpan
consistently and segfaulting, this patch increase
the data size to help improve stability.

Fixes #1255, #1408, #1415
commit e40cd69d (2017-07-28): [ocf] Improved OCF stability over 6lowpan (#1417)

The OCF was not able to send packets over BLE 6lowpan
consistently and segfaulting, this patch increase
the data size to help improve stability.

Fixes #1255, #1408, #1415