intel/zephyr.js #1256: native module 'ble' and 'ws' are not found

qiaojingx, 2017-06-16:
### Description

This is an regression issue. Require native module "ble" or "ws" got `NotSupportedError: module not found: ''` error.

### Test Code

[BLE.js](https://github.com/01org/zephyr.js/blob/master/samples/BLE.js); [WebSocketServer.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer.js) 
### Steps to Reproduction

1. make JS=samples/BLE.js ROM=256
2. make dfu
3. Reset 
### Actual Result

got `NotSupportedError: module not found: ''` error.
[image]
 
### Expected Result

BLE or websocket server started.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 44a5e6b | Arduino 101 | Jun 16, 2017 | Fail
master | a09bd64 | Arduino 101 | Jun 16, 2017 | Pass 
### Additional Information

grgustaf, 2017-06-16:
The empty module string is fixed by #1258; that's a step toward finding the real problem here.

qiaojingx, 2017-06-19:
Verified with commit 78c9749. The empty module string has been fixed. Native modules 'ble' and 'ws' still can't be found.
Platform: Arduino 101
[image]
[image]

grgustaf, 2017-06-20:
The fix is now merged, should work in 73a36db.

qiaojingx, 2017-06-21:
Verified with commit 15d3847. This issue is fixed. 'ble' and 'ws' module can be found. But another issue is found for websocket. new issue is here #1291 .
Close this one.

closed 2017-06-21 by qiaojingx