intel/zephyr.js #1257: [ashell] make ashell connect to WebIDE instead of terminal

qiaojingx, 2017-06-16:
### Description

This is an regression issue. `make ashell` should use the shell via the command line. But it connect to WebIDE, just like `make ide` did.

### Test Code

[ashell.md](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md); 
 
### Steps to Reproduction

1. make ashell ROM=256
2. make dfu
3. Reset 

### Actual Result

A notification from Chrome will appear with an URL of the IDE. A101 can be connect to WebIDE.
 
### Expected Result

It should use serial console instead of browser IDE. And user can use the following command to connect to the ashell from a terminal:
`$ screen /dev/ttyACM0 115200`
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 44a5e6b | Arduino 101 | Jun 16, 2017 | Fail
master | a09bd64 | Arduino 101 | Jun 16, 2017 | Pass 
### Additional Information

brianjjones, 2017-06-22:
zjs_ashell.json needs to be broken into multiple files to do this, I'll set that up.

closed 2017-06-30 by grgustaf

qiaojingx, 2017-07-04:
Verified with commit 418c871. This issue is fixed.

commit 418c8717 (2017-06-30): [ashell] Fix to allow command line ashell to be built (#1325)

Fixes #1257