intel/zephyr.js #1260: Recent updates broke build for most Zephyr boards

pfalcon, 2017-06-18:
Previously, (minimal) Zephyr.js could be built for any Zephyr board. Recently, build for any board besides handful explicitly "supported" is broken:

~~~
In file included from /mnt/hdd/projects-3rdparty/Embedded/Zephyr/zephyr.js/src/zjs_modules.c:21:0:
/mnt/hdd/projects-3rdparty/Embedded/Zephyr/zephyr.js/src/zjs_modules.c: In function ‘native_print_handler’:
/mnt/hdd/projects-3rdparty/Embedded/Zephyr/zephyr.js/src/zjs_util.h:39:23: error: implicit declaration of function ‘free’ [-Werror=implicit-function-declaration]
 #define zjs_free(ptr) free(ptr)
                       ^
/mnt/hdd/projects-3rdparty/Embedded/Zephyr/zephyr.js/src/zjs_modules.c:133:5: note: in expansion of macro ‘zjs_free’
     zjs_free(str);
     ^~~~~~~~
~~~

That's because Zephyr.js requires CONFIG_NEWLIB_LIBC, but instead of being defined globally for the application, it's now moved into individual zjs_qemu_x86.json, zjs_qemu_x86.json, etc.

I'm looking how to re-enable it globally.

pfalcon, 2017-06-18:
Ok, one might think that following would help:

~~~
zephyr.js/src$ cat zjs_all.json 
{
    "targets": ["all"],
    "zephyr_conf": {
        "all": [
            "CONFIG_STDOUT_CONSOLE=y",
            "CONFIG_NEWLIB_LIBC=y",
            "CONFIG_FLOAT=y",
            "CONFIG_NANO_TIMERS=y",
            "CONFIG_NANO_TIMEOUTS=y",
            "CONFIG_RING_BUFFER=y"
        ]
    }
}
~~~

But it doesn't. I'll leave it to you guys to see what's the best way to fix it.

jprestwo, 2017-06-19:
@pfalcon, this slipped my mind when implementing this. Either we need a prj.conf.base file (like we have for Makefile.base) or an "all" JSON file like you mentioned.

pfalcon, 2017-06-20:
@jprestwo : Thanks for looking into this. Zephyr already supports few dozens of boards, and the list is constantly growing. So, supporting any board by default, even if for small subset functionality, like console.log and GPIO, maybe a good idea, rather than not supporting by default. Adding more functionality for a board will certainly require creating a specific config, but doing that is much easier if the project is welcoming by providing some functionality OOB.

PR #1286 (2017-06-21): [analyze] Add zjs_common.json that always gets pulled in
This includes base configuration common to all platforms.

Fixes #1260.
commit 1d08c690 (2017-06-21): [analyze] Add zjs_common.json that always gets pulled in (#1286)

This includes base configuration common to all platforms.

Fixes #1260.
closed 2017-06-21 by jimmy-huang