intel/zephyr.js #1261: [net] The Socket.write() function will throw out error or be hang

cuiyanx, 2017-06-19:
### Description

The Socket.write() function will throw out error as client and be hang as server.

### Test Code

[/samples/TCPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ6.js)
[/samples/TCPClient6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPClient6.js)

 

### Steps to Reproduction

A101 as client:
1. Flash TCPClient6.js on a101
2. Connect by BLE
3. Run server on Linux
4. Look client logs

A101 as server:
1. Flash TCPEchoServ6.js on a101
2. Command `nc -6 2001:db8::1 4242` on Linux
3. Send `hello what haha`
4. Look server logs and be hang

 

### Actual Result

A101 logs as client:
```
receive data: Hello
send data: 'hello'
receive data: Hello

src/zjs_net.c:360 socket_write():
(ERROR) cannot acquire send_buf
send data: 'hello'

src/zjs_net.c:381 socket_write():
(ERROR) Cannot send data to peer (-128)
send data: 'hello'
socket connection error: WriteError

src/zjs_net.c:381 socket_write():
(ERROR) Cannot send data to peer (-128)
send data: 'hello'
```
A101 logs as server:
```
connection made
port: 4242
family: IPv6
address: a00:1092::
socket.localPort: 4242
socket.localAddress: a00:1092::
socket.remotePort: 4242
socket.remoteAddress: a00:8016:fd50:483d:73e2::
socket.remoteFamily: IPv6
Got data: hello

Got data: what

Got data: haha
```
Linux logs as client:
 ```
root@intelProject:/home/cuiyan/intelwork/cy/zephyr.js/tests/tools# nc -6 2001:db8::1 4242
hello
hello
what
what
haha
^C
```
 

### Expected Result

Work well.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master |  78c9749 | Arduino 101 | June 19, 2017 | Fail

 

### Additional Information

grgustaf, 2017-07-26:
I don't see this with ENC28J60 ethernet board, so similar to #1188 I think the priority can be lowered. These seem like the same bug now at least.

cuiyanx, 2017-07-27:
@grgustaf 
As server on a101: still be hang
As client on a101:
Simple test as client (will flash a101):
```
var net = require('net');
var net_cfg = require('net-config');

net_cfg.on('netup', function() {
	console.log("network up");
	setTimeout(function() {
		console.log("starting client");
		var client = new net.Socket();

		var count = 0;
		var is_connected = false;

		function connected() {
			console.log("connect successful");
			is_connected = true;
			client.write(new Buffer("initial data\r\n"));
		}

		client.on("data", function(data) {
			count++;
			console.log("got data: " + data.toString('ascii'));
			client.write(new Buffer("write data " + count + "\r\n"));
		});
		client.on("close", function() {
			console.log("Socket has closed");
		});
		client.on("error", function(err) {
			if (err.name == "NotFoundError") {
				console.log("Server not found, retrying in 2 seconds");
				setTimeout(function() {
					if (!is_connected) {
						client.connect({port: 4242, host:'2001:db8::2', localAddress:'2001:db8::1', localPort: 8484, family: 6}, connected);
					}
				}, 2000);
			}
		});

		console.log("Connecting");
		client.connect({port: 4242, host:'2001:db8::2', localAddress:'2001:db8::1', localPort: 8484, family: 6}, connected);
	}, 10000);
});
```
Simple test server (run on Linux by python):
```
# !usr/bin/python
# coding:utf-8

import time
import socket
import SocketServer

def main():
    print "Socket server creat successful"

    host = "2001:db8::2"
    port = 4242
    addr = (host, port)
    Timeout = 80
    bufSize = 1024

    mySocket = socket.socket(socket.AF_INET6, socket.SOCK_STREAM)
    mySocket.settimeout(Timeout)
    mySocket.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
    mySocket.bind(addr)
    mySocket.listen(5)

    client, clientaddr = mySocket.accept()
    print "Connected from: ", client.getpeername()

    while 1:
        try :
            Data = mySocket.recv(bufSize)
            Data = Data.strip()
            print "Got data: ", Data

            time.sleep(2)

            if Data == "close":
                mySocket.close()
                print "close socket"

                break
            else:
                mySocket.sendall(Data)
                print "Send data: ", Data
        except :
            print "time out"
            continue

if __name__ == "__main__" :
    main()

```
Step:
1. Build and flash this client on a101.
2. Connect a101 with ble.
3. Add IPv6 route and IPv6 address.
4. Run this server on Linux by python.
5. Check console log.

Log:
```
Connected
network up
starting client
Connecting

src/zjs_net.c:916 socket_connect():
(ERROR) Error in net_context_bind(handle->tcp_sock, (struct sockaddr *)&my2
Error: net_context_bind(handle->tcp_sock, (struct sockaddr *)&my_addr6, si)
```

Looks like #1413.

cuiyanx, 2017-08-03:
Fixed and closed.

closed 2017-08-03 by cuiyanx