intel/zephyr.js #1262: [ashell]Failed to run zephyrjs-ide examples after updating zephyr.js

qiuzhong, 2017-06-19:
## Description:
After I update zephyr.js to [44a5e6b](https://github.com/01org/zephyr.js/commit/44a5e6b8547af30583cc324d1263464e4fd76325), the [zephyrjs-ide](https://github.com/01org/zephyrjs-ide) samples are not working any more.

## Steps to Reproduce
1. Update the  [zephyr.js](https://github.com/01org/zephyr.js) to [44a5e6b](https://github.com/01org/zephyr.js/commit/44a5e6b8547af30583cc324d1263464e4fd76325), and update its submodules.
2. Build the **ide** image and flash it.
3. Launch zephyrjs-ide and run the examples.

## Actual Result:
`Acelerometer / Gyroscope / Pattern Matching / Sensor BLE Demo` examples don't work.
Here are the errors
```
acm> BMI160 accelerometer test...
Error: Accelerometer is not defined

acm> BMI160 gyroscope test...
Error: Gyroscope is not defined

acm> Curie Pattern Matching Engine test...
In function require():
NotSupportedError: module not found: 'pme'
```

## Expected Result:
All the examples can still work.

## Test Builds
| Branch | Commit Id | Target Device | Test Date | Result |
| --- | --- | --- | --- | --- |
| master |	[8d1b343](https://github.com/01org/zephyr.js/commit/8d1b343079d2a03700ab623281acb8b6ea5e1541) | Arduino 101 | June 9, 2017 |	Pass |
| master |	[44a5e6b](https://github.com/01org/zephyr.js/commit/44a5e6b8547af30583cc324d1263464e4fd76325) | Arduino 101 | June 16, 2017 |	Failed |

grgustaf, 2017-06-21:
Oh, this is probably mostly because sensor got dropped from the IDE image. That should have been fixed by #1253. But sensor may still not be working based on my testing and comments from that PR, so @jimmy-huang please try it.

qiaojingx, 2017-06-22:
Verified with commit e590c30. Acelerometer / Gyroscope / Sensor BLE Demo works fine in WebIDE. But Sample(Pattern Matching) still got error `NotSupportedError: module not found: 'pme'`

jimmy-huang, 2017-06-27:
@qiaojingx PME is not dropped from ide because they don't fit any more, so it's intended.

qiuzhong, 2017-06-28:
IDE examples need to update, see [ZJS-IDE issue#173](https://github.com/01org/zephyrjs-ide/issues/173).  So close this issue.

closed 2017-06-28 by qiuzhong