intel/zephyr.js #1263: SPI API

poussa, 2017-06-19:


qiaojingx, 2017-06-30:
Basic feature of SPI is ready on Arduino 101 and FRDM-K64F. Verified with samples([SPI.js](https://github.com/01org/zephyr.js/blob/master/samples/SPI.js)). One known issue is on FRDM-K64F(#1326).
Commit: c2ca034
Platform: Arduino 101 and FRDM-K64F.

grgustaf, 2017-07-06:
Added by @brianjjones

closed 2017-07-06 by grgustaf

qiaojingx, 2017-07-07:
Basic feature of SPI is ready on Arduino 101 and FRDM-K64F. Verified with samples([SPI.js](https://github.com/01org/zephyr.js/blob/master/samples/SPI.js)). 1 known issue(#1326) is on FRDM-K64F, which is possibly a zephyr issue(tracking by https://jira.zephyrproject.org/browse/ZEP-2353).
Commit ID: 0334065
Platform: Arduino 101 and FRDM-K64F.
ZephyrSDK: 0.9.1
Zephyr:1.8