intel/zephyr.js #1264: Node.js TCP (net) API

poussa, 2017-06-19:
https://nodejs.org/api/net.html

qiaojingx, 2017-06-30:
Basic feature of TCP(Net) is ready on Arduino 101. But blocked by #1148 on FRDM-K64F. Verified with samples([TCPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ6.js), [TCPClient6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPClient6.js)) and tests([test-tcp6-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp6-server.js), [test-tcp6-client.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp6-client.js)). 8 known issues for TCP(net) are: #1102, #1172, #1175, #1187, #1188, #1218, #1261, #1319.

grgustaf, 2017-07-06:
Added by @jprestwo

closed 2017-07-06 by grgustaf

qiaojingx, 2017-07-07:
Basic feature of TCP(Net) is ready on both Arduino 101 and FRDM-K64F. Verified with samples([TCPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ6.js), [TCPClient6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPClient6.js)) and tests([test-tcp6-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp6-server.js), [test-tcp6-client.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp6-client.js)). 8 known issues for TCP(net) are: #1102, #1172, #1175, #1187, #1188, #1218, #1261, #1319.
Commit ID: 0334065
Platform: Arduino 101 and FRDM-K64F.
ZephyrSDK: 0.9.1
Zephyr:1.8