intel/zephyr.js #1265: JavaScript Module Support ashell

poussa, 2017-06-19:
Add support for JavaScript modules with ashell and Web IDE

qiaojingx, 2017-07-05:
JS modules can be required in ashell and WebIDE. Verified with JS modules([Assert.js](https://github.com/01org/zephyr.js/blob/master/modules/Assert.js), [Driver.js](https://github.com/thewebera/zjs-demos/blob/master/minicar/modules/Driver.js), [Steerer.js](https://github.com/thewebera/zjs-demos/blob/master/minicar/modules/Steerer.js))
1 know issue is found: #1226 .
Commit: 418c871
Platform: Arduino 101
WebIDE: v0.18.3-alpha / 502fccc

brianjjones, 2017-07-06:
This should be ready to go.

qiaojingx, 2017-07-17:
JS modules can be required in ashell and WebIDE. Verified with JS modules([Assert.js](https://github.com/01org/zephyr.js/blob/master/modules/Assert.js)).
Commit: af7eadd
Platform: Arduino 101
ZephyrSDK: 0.9.1
Zephyr:1.8
WebIDE: v0.18.3-alpha / 502fccc

closed 2017-08-14 by grgustaf