intel/zephyr.js #1266: Web Socket API

poussa, 2017-06-19:
Subset of WS API : https://html.spec.whatwg.org/multipage/web-sockets.html

grgustaf, 2017-07-06:
Added by @jprestwo

closed 2017-07-06 by grgustaf

qiaojingx, 2017-07-20:
Basic feature is ready on FRDM-K64F. But there is still a block issue(#1348) on Arduino101. Verified with samples([WebSocketServer.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer.js), [WebSocketServer4.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer4.js)) and tests([test-ws-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ws-server.js)). 2 other issues are found : #1234, #1369.
Commit ID: de91ade + #1374 
Platform: Arduino 101 and FRDM-K64F.
ZephyrSDK: 0.9.1
Zephyr:1.8

qiaojingx, 2017-07-21:
Basic feature is ready on Arduino101 and FRDM-K64F. Verified with samples([WebSocketServer.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer.js), [WebSocketServer4.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer4.js)) and tests([test-ws-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ws-server.js)). 4 issues are found : #1234, #1369, #1385, #1386. 
Commit ID: 5c65f11
Platform: Arduino 101 and FRDM-K64F.
ZephyrSDK: 0.9.1
Zephyr:1.8