intel/zephyr.js #1267: Support for multiple BLE connections

poussa, 2017-06-19:
Add support for multiple BLE connections. Due to memory limitations max number of connections is 4.

qiaojingx, 2017-06-28:
Multiple BLE connections is supported. Verified with samples([BLE.js](https://github.com/01org/zephyr.js/blob/master/samples/BLE.js)) and tests([test-ble.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ble.js)). It can accept 4 connections at the same time.
Commit: 8a95b21
Platform: Arduino 101.

grgustaf, 2017-07-06:
Added by @jimmy-huang

closed 2017-07-06 by grgustaf

qiaojingx, 2017-07-07:
Multiple BLE connections is supported. Verified with samples([BLE.js](https://github.com/01org/zephyr.js/blob/master/samples/BLE.js)) and tests([test-ble.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ble.js)). 4 connections can be accepted.
Commit ID: 0334065
Platform: Arduino 101. 
ZephyrSDK: 0.9.1