intel/zephyr.js #1270: [gpio] Failed to build with gpio module for K64F

qiaojingx, 2017-06-19:
### Description

This is an regression issue. Failed to build samples which required native module 'gpio' for K64F.

### Test Code

[Blink-k64f.js](https://github.com/01org/zephyr.js/blob/master/samples/arduino/basics/Blink-k64f.js); [ButtonLEDs-k64f.js](https://github.com/01org/zephyr.js/blob/master/samples/ButtonLEDs-k64f.js); [GPIOOutputs-k64f.js](https://github.com/01org/zephyr.js/blob/master/samples/tests/GPIOOutputs-k64f.js); [GPIOInputs-k64f.js](https://github.com/01org/zephyr.js/blob/master/samples/tests/GPIOInutputs-k64f.js); [test-k64f-pins.js](https://github.com/01org/zephyr.js/blob/master/tests/test-k64f-pins.js) 
### Steps to Reproduction

1. `make BOARD=frdm_k64f JS=samples/ButtonLEDs-k64f.js`
2. `cp outdir/frdm_k64f/zephyr.bin /media/<username>/MBED/`

### Actual Result

Build failed
[image]

 ### Expected Result

sample build successfully.

 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 78c9749 | FRDM-K64F | Jun 19, 2017 | Fail
master | a09bd64 | FRDM-K64F | Jun 19, 2017 | Pass 

### Additional Information

grgustaf, 2017-06-22:
Working in commit ff8e80d, no thanks to me. :)

qiaojingx, 2017-06-23:
Verified with commit 8a95b21, this issue has been fixed.
Device: FRDM-K64F.
Closed

closed 2017-06-23 by qiaojingx