intel/zephyr.js #1271: Pattern Matching API

poussa, 2017-06-19:
Add basic API to access Curie's pattern matching engine (PME)

qiaojingx, 2017-06-28:
Basic feature of PME is ready on Arduino 101. Verified with samples([PME.js](https://github.com/01org/zephyr.js/blob/master/samples/PME.js) , [PMESaveRestore.js ](https://github.com/01org/zephyr.js/blob/master/samples/PMESaveRestore.js )) and tests([test-pme.js](https://github.com/01org/zephyr.js/blob/master/tests/test-pme.js)). 6 known issues are:  #1243, #1244, #1245, #1246, #1249, #1254.
Commit ID: 8a95b21
Platform: Arduino 101.

grgustaf, 2017-07-06:
Added by @jimmy-huang

closed 2017-07-06 by grgustaf

qiaojingx, 2017-07-07:
Basic feature of PME is ready on Arduino 101. Verified with samples([PME.js](https://github.com/01org/zephyr.js/blob/master/samples/PME.js) , [PMESaveRestore.js ](https://github.com/01org/zephyr.js/blob/master/samples/PMESaveRestore.js )) and tests([test-pme.js](https://github.com/01org/zephyr.js/blob/master/tests/test-pme.js)). 1 known issue is: #1254.
Commit ID: 0334065
Platform: Arduino 101. 
ZephyrSDK: 0.9.1