intel/zephyr.js #1279: [uart] Uart functions are broken

cuiyanx, 2017-06-20:
### Description

I test those function:
```
void write(Buffer data)
void setReadRange(number min, number max)
function onread(Buffer data)
```
I always get `Error: Expected a function`. Is this a bug or functions to be completed?

### Test Code

[/samples/UART.js](https://github.com/01org/zephyr.js/blob/master/samples/UART.js)

 

### Steps to Reproduction

 

### Actual Result

 Throw out error.

### Expected Result

 Work well.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 78c9749 | Arduino 101 | June 20, 2017 | Fail

 

### Additional Information

brianjjones, 2017-06-23:
This is fixed by #1310

closed 2017-06-29 by grgustaf

qiaojingx, 2017-06-30:
Verified with commit c2ca034, this issue is not reproducible.

cuiyanx, 2018-01-10:
Fixed and closed

commit a2d6607c (2017-06-29): [uart] Fix broken UART apis (#1310)

The lock in signal_callback is not freeing before UART is
triggered.  This is causing UART to be locked out when it tries
to write.  Moving the unlock up fixes the issue.  Also adding a
needed item to the zjs_uart.json file.

Fixes #1279