intel/zephyr.js #1280: [SPI] "Hello World\0" is identified as invalid ascii

qiaojingx, 2017-06-20:
### Description

When running [SPI.js](https://github.com/01org/zephyr.js/blob/master/samples/SPI.js); on Arduino 101. "Hello World\0" is identified as invalid ascii

### Test Code

[SPI.js](https://github.com/01org/zephyr.js/blob/master/samples/SPI.js); 

### Steps to Reproduction

1. add console.log to SPI.js as below:
     
[image]

2. `make JS=samples/SPI.js`
3. `make dfu`

### Actual Result

Error `SPI error: buffer has invalid ascii ` is print.
[image]

 ### Expected Result

Running SPI sample successfully.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 78c9749 | Arduino 101 | Jun 20, 2017 | Fail

### Additional Information

When running [SPI.js](https://github.com/01org/zephyr.js/blob/master/samples/SPI.js); on K64F. The below error is print.
[image]
Is that SPI module supported on K64F, too?

grgustaf, 2017-06-21:
Brian things you didn't have the MOSI and MISO lines connected with a wire. That makes the "read" come in as all 1's, which triggers that message when we try to convert to ASCII.

Really, we can't be guaranteed that we'll get ASCII back from a SPI read so makes more sense to fix this by printing the buffer in hex.

This does point out an interesting fact about JS strings - they can contain NULL bytes, which is useful for specifying buffers like this as strings in a compact form. So e.g. the size of the "Hello World\0" string above is (I think) 12 bytes, containing a 0 at the end.

qiaojingx, 2017-06-27:
@brianjjones, this issue has been fixed on Arduino101, but still failed on K64F. Does it support on k64f, or only for Arduino101?
Result:
Arduino101:
[image]
K64F:
[image]
Commit ID: 8a95b21

brianjjones, 2017-06-27:
Currently its only tested for A101.  I'll start looking into K64 this week

brianjjones, 2017-06-28:
@qiaojingx  I've updated things so that it 'works' on k64f https://github.com/01org/zephyr.js/pull/1323

However, I'm only seeing zero's in the reply buffer.  Currently I'm not sure if it's an issue with Zephyr, the board itself, or the current example's configuration.  I'll keep looking into it and let you know if I find a solution.

qiaojingx, 2017-06-30:
Verified with commit c2ca034 on both Arduino101 and FRDM-K64F. This issue is not reproducible.
Created a new issue for only seeing zero's in the reply buffer on K64F( #1326 ).
Close this one.

closed 2017-06-30 by qiaojingx