intel/zephyr.js #1283: [modules] require customer module(GroveLCD.js/BMP280.js ) failed on Arduino101

qiaojingx, 2017-06-20:
### Description

This is an regression issue. When try to require customer module [GroveLCD.js](https://github.com/01org/zephyr.js/blob/master/modules/GroveLCD.js) or [BMP280.js](https://github.com/01org/zephyr.js/blob/master/modules/BMP280.js) on Arduino101. It failed and nothing will output in console.log. But this issue is not reproduced on K64F.

### Test Code

[I2CBMP280.js](https://github.com/01org/zephyr.js/blob/master/samples/I2CBMP280.js)

### Steps to Reproduction

1. Build with the following code for Arduino 101.
```
console.log('--------------1----------');
var bmp280 = require("BMP280.js");
console.log('--------------2----------');
```
### Actual Result

Nothing is outputted in console.log.

 ### Expected Result

Samples should be executed successfully.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 78c9749 | Arduino 101 | Jun 20, 2017 | Fail
master | a09bd64 | Arduino 101 | Jun 20, 2017 | Pass

### Additional Information

require customer module [Assert.js](https://github.com/01org/zephyr.js/blob/master/modules/Assert.js) succeed on Arduino 101.

brianjjones, 2017-06-21:
This is fixed by https://github.com/01org/zephyr.js/pull/1287

qiaojingx, 2017-06-23:
Verified with commit 8a95b21. This issue has been fixed.
Device: Arduino 101.
Closed

closed 2017-06-23 by qiaojingx