intel/zephyr.js #1288: [fs] Please complete FS APIs on fs.md

cuiyanx, 2017-06-21:
### Description

I find FS APIs can not be completed, there are just only one interface for state.
And there is no function for [/samples/FsAsync.js](https://github.com/01org/zephyr.js/blob/master/samples/FsAsync.js). Will we add it? If not, please remove it.
 

### Test Code

[/docs/fs.md](https://github.com/01org/zephyr.js/blob/master/docs/fs.md)
 [/samples/FsAsync.js](https://github.com/01org/zephyr.js/blob/master/samples/FsAsync.js)

 

### Steps to Reproduction

 

### Actual Result

Not be completed.
 

### Expected Result

Be finished.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 15d38476 | Arduino 101 | June 21, 2017 | Fail

 

### Additional Information

grgustaf, 2017-06-21:
I think this is saying the documentation is incomplete. The IDL portion doesn't include most of the APIs, and the Async APIs are left out.

The latter bit was intentional - we weren't sure if we really intend to support them; they are there for backwards compatibility but not really meant to be used for new code.

qiaojingx, 2017-06-23:
@grgustaf ,  [FsAync.js](https://github.com/01org/zephyr.js/blob/master/samples/FsAsync.js) sample can't be executed successfully on A101 now.
[image]

grgustaf, 2018-01-24:
"there are just only one interface for state"
@cuiyanx, can you tell me what that meant? The missing async docs is not high-priority, since that's intentional but I wonder if this is more important?

cuiyanx, 2018-01-25:
@grgustaf We had some APIs for synchronous situation and none for asynchronous situation. But we have sample for asynchronous situation. We maybe need to add APIs for asynchronous situation, or remove this sample. I think that is my thought.