intel/zephyr.js #1289: [ocf] ipm timed out error occurs when running ocf sensor server.

qiaojingx, 2017-06-21:
### Description

`FATAL ERROR: ipm timed out` error occurs when running [OcfSensorServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfSensorServer.js)(or [OcfMultiSensor.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfMultiSensor.js)) on Arduino 101 and [OcfSensorClient.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfSensorClient.js) on Linux

### Test Code

[OcfSensorServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfSensorServer.js); [OcfMultiSensor.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfMultiSensor.js);

### Steps to Reproduction

1. Build [OcfSensorServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfSensorServer.js) on A101 via BLE.
2. Build [OcfSensorClient.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfSensorClient.js) on Linux.

### Actual Result

`FATAL ERROR: ipm timed out` error occurs
OcfSensorServer.js
[image]

OcfMultiSensor.js
[image]

OcfSensorClient.js
[image]

 ### Expected Result

OcfSensorClient can receive sensor value which measured on OcfSensorServer.

 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 78c9749 | Arduino101 | Jun 21, 2017 | Fail
master | a09bd64 | Arduino101 | Jun 20, 2017 | Pass

### Additional Information

jimmy-huang, 2017-06-26:
I can't reproduce this on master, with OcfSensorServer.js I don't see the IPM time out error:
Started OCF server
Registered resource
Connected
oc_network_receive: received 53 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe78:1d72]:48337
oc_send_buffer: outgoing message: [fe80:0000:0000:0000:5ef3:70ff:fe78:1d72]:48337
oc_send_buffer: failed: (38)
oc_network_receive: received 25 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe78:1d72]:48337
ocf_get_handler()
on('retrieve'): request.target.resourcePath=/a/sensor observe=false
oc_send_buffer: outgoing message: [fe80:0000:0000:0000:5ef3:70ff:fe78:1d72]:48337
oc_send_buffer: sent 34 bytes
respond success
oc_network_receive: received 25 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe78:1d72]:48337
ocf_get_handler()
on('retrieve'): request.target.resourcePath=/a/sensor observe=false
oc_send_buffer: outgoing message: [fe80:0000:0000:0000:5ef3:70ff:fe78:1d72]:48337
oc_send_buffer: sent 34 bytes
respond success

But I see the OcfClient segfault, and when running OcfMultiSensor.js, it seems the client is stuck after issuing the first request, is this what you see?

qiaojingx, 2017-06-27:
@jimmy-huang , ` ipm timed out` issue is not reproduced with the latest commit 8a95b21 in master. I can reproduce segmentation fault when running OcfSensorClient.js(tracking #1316 ). But didn't see the client stuck issue when running OcfMultiSensor.js.
Close this one.

closed 2017-06-27 by qiaojingx