intel/zephyr.js #1293: [spi] Please add SPI bus information

cuiyanx, 2017-06-21:
### Description

After I create SPI bus, I can not get SPI bus information.

In [/docs/spi.md](https://github.com/01org/zephyr.js/blob/master/docs/spi.md):
```
dictionary SPIOptions {
    octet bus;
    long speed;
    bool msbFirst;
    long polarity;
    long phase;
    unsigned long frameGap;
    string topology;

};
```
 

### Test Code

Simple test:

```
var spi = require("spi");

var spiBus = spi.open({speed:20, bus:1, polarity:0, phase:0, bits:16});

var spiBusOptions = [
    [spiBus.bus, "bus"],
    [spiBus.speed, "speed"],
    [spiBus.msbFirst, "msbFirst"],
    [spiBus.polarity, "polarity"],
    [spiBus.phase, "phase"],
    [spiBus.frameGap, "frameGap"],
    [spiBus.topology, "topology"]
];

for (var i = 0; i < spiBusOptions.length; i++) {
    console.log("spiBus." + spiBusOptions[i][1] + ": " + spiBusOptions[i][0]);
}

```
 

### Steps to Reproduction

 

### Actual Result

Logs:

```
spiBus.bus: undefined
spiBus.speed: undefined
spiBus.msbFirst: undefined
spiBus.polarity: undefined
spiBus.phase: undefined
spiBus.frameGap: undefined
spiBus.topology: undefined
```

 

### Expected Result

 Can get SPI bus information.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 15d3847 | Arduino 101 | June 21, 2017 | Fail

 

### Additional Information

brianjjones, 2017-06-21:
These options are all read only, the user should be aware what they set them to.  Adding them will increase the size of the object, and I'm not sure if we wanted to do that.  @zolkis thoughts?

grgustaf, 2017-06-21:
Yeah, I was going to say the same thing basically. It matches the documentation as it is - the SPIBus interface does not claim to have those properties.

cuiyanx, 2017-06-22:
OK, I refer to this document [/board/spi.md](https://github.com/zolkis/iot-js-api/blob/master/board/spi.md).
Closed.

closed 2017-06-22 by cuiyanx