intel/zephyr.js #1294: [spi] SPI bus can only read and write array and string

cuiyanx, 2017-06-21:
### Description

Now, SPI bus can only read and write array and string. I test buffer data as `number` and `boolean`, SPI bus will be hang.
 

### Test Code

Change `"Hello World\0"` to `true` or `1024` in [/samples/SPI.js](https://github.com/01org/zephyr.js/blob/master/samples/SPI.js) on line 10.

 

### Steps to Reproduction

 

### Actual Result

Logs: `SPI test starting..`
 

### Expected Result

 Can read and write number or boolean or some buffer data.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 15d3847 | Arduino 101 | June 21, 2017 | Fail

 

### Additional Information

brianjjones, 2017-06-21:
Currently numbers and bools are not supported.  But I don't see why they couldn't be so I'll look into adding it.

grgustaf, 2017-06-21:
I really don't think that makes sense. Accepting Buffer, however, makes a lot of sense. :)

grgustaf, 2017-06-21:
Marking P2, but this is for handling Buffer.

brianjjones, 2017-06-27:
https://github.com/01org/zephyr.js/pull/1315 will add the ability to pass it a Buffer

qiaojingx, 2017-06-29:
Verified with commit 3403f93. Buffer object can be passed.

cuiyanx, 2017-07-10:
fixed and closed

closed 2017-07-10 by cuiyanx