intel/zephyr.js #1301: [analyze] Can not found modules after adding zjs_common.json

cuiyanx, 2017-06-22:
### Description

 After adding zjs_common.json, I can not found modules.

### Test Code

[/tests/test-a101-pins.js](https://github.com/01org/zephyr.js/blob/master/tests/test-a101-pins.js)
[/tests/test-console-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-console-manual.js)
[/tests/test-pme.js](https://github.com/01org/zephyr.js/blob/master/tests/test-pme.js)

 

### Steps to Reproduction

 

### Actual Result

 Error logs:

`Error: console is not defined` 
or `NotSupportedError: module not found: 'performance'` 
or `NotSupportedError: module not found: 'arduino101_pins'`

### Expected Result

 Work well.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 1f74fb926 | Arduino 101 | June 22, 2017 | Pass
master | 1d08c690 | Arduino 101 | June 22, 2017 | Fail

 

### Additional Information

grgustaf, 2017-06-22:
This was fixed by PR #1300; sorry we should have gotten that in last night.

qiaojingx, 2017-06-23:
Verified with commit 8a95b21f. This issue has been fixed.
Device: Arduino 101.

cuiyanx, 2017-07-10:
fixed and closed

closed 2017-07-10 by cuiyanx