intel/zephyr.js #1316: [ocf] Segmentation fault occurs when running ocf client on Linux

qiaojingx, 2017-06-27:
### Description

When running [OcfSensorClient.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfSensorClient.js) and [test-ocfclient.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ocfclient.js) on Linux, `Segmentation fault (core dumped)` error occurs.

### Test Code

[OcfSensorClient.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfSensorClient.js);
[OcfSensorServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfSensorServer.js);
[test-ocfserver.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ocfserver.js);
[test-ocfclient.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ocfclient.js)
 
### Steps to Reproduction

Scenario 1:
1. Build [OcfSensorServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfSensorServer.js) on Arduino101 via BLE.
2. Build [OcfSensorClient.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfSensorClient.js) on Linux

Scenario 2:
1. Running [test-ocfserver.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ocfserver.js) in terminal 1 on Linux
2. Running [test-ocfclient.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ocfclient.js) in terminal 2 on Linux

### Actual Result

Scenario 1 and 2 both got `Segmentation fault (core dumped)` error.
Scenario 1:
Client on Linux:
[image]
Server on A101:
[image]
Scenario 2:
Client on Linux:
[image]

Server on Linux:
[image]

### Expected Result

No error occurs.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 8a95b21 | Linux | Jun 27, 2017 | Fail
 
### Additional Information

jimmy-huang, 2017-06-28:
It's crashing on
client.update(resource).then(function(res) {
}).catch(function(error) {
});

And the callstack is showing it crashing on executing the callback in the job queues for promises.
For now, it appears to segfault only when the function passed into the .then() function has a empty body, so a work-around is to add a single statement like console.log() or ASSERT() in the .then(function(res) {
}).

It might be a upstream JerryScript bug, maybe @jprestwo have some insights on this one?

jprestwo, 2017-06-28:
A cleaner workaround would be to just not call `.then()` or `.catch()` at all.

```
client.update(resource);
```

qiaojingx, 2017-07-06:
Verified with 4e44cc6 + #1332 . `Segmentation fault (core dumped)` error is not reproduced when running [test-ocfclient.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ocfclient.js). But it still occurs when running [OcfSensorClient.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfSensorClient.js);
[image]

qiaojingx, 2017-07-13:
Verified with commit 1ff882b + #1332 . `Segmentation fault (core dumped)` error is not reproducible. 
But, "update" events didn't get called on ocf client.
[image]

jimmy-huang, 2017-07-14:
@qiaojingx, @grgustaf and I discussed a little bit on the API, we agree that the update() events should only be called when the resource has changed, so it doesn't make sense for the events to be called each time you call retrieve(), does that make sense?

closed 2017-07-14 by grgustaf

qiaojingx, 2017-07-17:
Verified with commit af7eadd. This issue is not reproduced.

PR #1332 (2017-07-14): [ocf] Work-around to fix segfault when using promises
This is a work-around patch to have the tests no segfault
when the function passed into the .then() is empty, which
causes a segfault executed.

Fixes #1316
commit 007433a1 (2017-07-14): [ocf] Changed "update" events to only get called when changed

Instead of triggering onchange event everytime retrieve() is
called, the client should only trigger when a actual resource
has changed from the server.

Fixes #1316
commit de6bb66b (2017-07-14): [ocf] Work-around to fix OCF segfault

There's a possible race condition in JerrScript or ZJS that parameters
were getting freed when promises are resolved and at the same time
the same jerry value is passed to another function through trigger
event.  This is a work-around patch to fix this until the root cause
is found.

Fixes #1316