intel/zephyr.js #1317: [ashell] Memory malloc failed

qiuzhong, 2017-06-27:
## Description
I encountered this memory `malloc failed` error when running the IDE examples occasionally. 
```
acm> ashell/jerry-code.c:157 read_file_alloc():
(ERROR) malloc failed
```
The example I was using is the `BLE`, this is not a large size file (2180 Bytes). There is no other file via the ashell `ls` command.

## How  did I encounter this?
**Notice**: the following steps are not guaranteed to reproduce it 100%. 
1. Build `ide` image with `ROM=256`
2. Launch IDE and run the examples and stop them randomly.

## Build
| Branch |	Commit Id | Target Device | 
| --- | --- | --- | 
| master | [8a95b21](https://github.com/01org/zephyr.js/commit/8a95b21fee63130911cd94e512f2819f973f9794) | Arduino 101 |

qiuzhong, 2017-06-27:
More logs:
```
acm> BLE sample...

acm> stop

acm> Starting Blink sample...

acm> stop

acm> BMI160 gyroscope test...
src/zjs_callbacks.c:490 zjs_add_c_callback():
(ERROR) malloc failed
src/zjs_callbacks.c:490 zjs_add_c_callback():
(ERROR) malloc failed
src/zjs_callbacks.c:490 zjs_add_c_callback():
(ERROR) malloc failed
src/zjs_callbacks.c:331 add_callback_priv():
(ERROR) malloc failed
ashell/comms-shell.c:412 ashell_process_data():
(ERROR) malloc failed
ashell/comms-shell.c:412 ashell_process_data():
(ERROR) malloc failed
```

brianjjones, 2017-08-05:
There are likely memory leaks in a few of the modules, possibly ashell as well.  To track them down build ashell with TRACE=on and you'll see a printout of all the remaining memory when you stop a JS file.  After multiple times it should be clear if there is memory not being freed, and where that memory got allocated in the code.

canxcao, 2017-08-10:
Verified issue with commit id 4f773db, it cannot reproduce.
 Close.

jimmy-huang, 2017-08-14:
@qiuzhong can you verify if this bug is still reproducible? Please close it if it is fixed.

qiuzhong, 2017-08-15:
Not reproduced. So close it.

closed 2017-08-15 by qiuzhong