intel/zephyr.js #1320: W3C Sensor support for FRDM-K64F

jimmy-huang, 2017-06-28:
W3C Sensors should support on-board sensors on the FRDM-K64F like Accelerometer and Magnetometer

cuiyanx, 2017-07-20:
Basic feature is ready on FRDM-K64F.  Verified with samples([FXOS8700Accelerometer.js](https://github.com/01org/zephyr.js/blob/master/samples/FXOS8700Accelerometer.js), [FXOS8700Magnetometer.js](https://github.com/01org/zephyr.js/blob/master/samples/FXOS8700Magnetometer.js)) and tests([test-bmi160accelerometer.js](https://github.com/01org/zephyr.js/blob/master/tests/test-bmi160accelerometer.js)).
Commit ID: de91ade
Platform: FRDM-K64F.
ZephyrSDK: 0.9.1
Zephyr:1.8

jimmy-huang, 2017-08-02:
closing this as it's implemented

closed 2017-08-02 by jimmy-huang

qiaojingx, 2017-08-04:
@jimmy-huang ，Other two W3C sensors( [BMI160Gyroscope.js](https://github.com/01org/zephyr.js/blob/master/samples/BMI160Gyroscope.js)，[BMI160Temperature.js](https://github.com/01org/zephyr.js/blob/master/samples/BMI160Temperature.js) ) are not support on K64F, do they behave as expected ?