intel/zephyr.js #1326: [SPI] There are only zero's in the reply buffer on K64F

qiaojingx, 2017-06-30:
### Description

 When running [SPI.js](https://github.com/01org/zephyr.js/blob/master/samples/SPI.js) on FRDM-K64F, only seeing zero's in the reply buffer.

### Test Code

[SPI.js](https://github.com/01org/zephyr.js/blob/master/samples/SPI.js);
 
### Steps to Reproduction

 1. `make JS=samples/SPI.js BOARD=frdm_k64f`
 2. `cp outdir/frdm_k64f/zephyr.bin /media/<username>/MBED/`

### Actual Result

 only seeing zero's in the reply buffer:
[image]
### Expected Result

Reply content is correct. 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | c2ca034 | FRDM-K64F | Jun 30, 2017 | Fail
 
### Additional Information

brianjjones, 2017-06-30:
This is possibly a Zephyr issue, I've opened https://jira.zephyrproject.org/browse/ZEP-2353 and will work with them to find out if that's the case.

brianjjones, 2017-07-19:
This issue was due to the fact that K64f uses SPI_0 not SPI_1 for communication, I updated the example to check the board and use the correct SPI bus.

closed 2017-07-24 by grgustaf

qiaojingx, 2017-07-25:
Verified with commit 5726f4e. This issue is fixed.

commit bac59771 (2017-07-24): [spi] Make SPI sample work with k64f (#1380)

Fixes #1326