intel/zephyr.js #1327: [Sensor] garbled code in magnetometer output

qiaojingx, 2017-06-30:
### Description

 When running [FXOS8700Magnetometer.js](https://github.com/01org/zephyr.js/blob/master/samples/FXOS8700Magnetometer.js) on FRDM-K64F, there is garbled code in magnetometer output.

### Test Code

[FXOS8700Magnetometer.js](https://github.com/01org/zephyr.js/blob/master/samples/FXOS8700Magnetometer.js)
 
### Steps to Reproduction

 1. `make JS=samples/FXOS8700Magnetometer.js BOARD=frdm_k64f`
 2. `cp outdir/frdm_k64f/zephyr.bin /media/<username>/MBED/`

### Actual Result

there is messy code outputted:
[image]

### Expected Result

No garbled code.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | c2ca034 | FRDM-K64F | Jun 30, 2017 | Fail
 
### Additional Information

jimmy-huang, 2017-08-07:
I don't think I can reproduce this with master, can you verify?

qiaojingx, 2017-08-08:
Verified with commit 9a272ee on FRDM-K64F.
This issue can be reproduced by `minicom -D /dev/ttyACM0`, but not reproduced by `screen /dev/ttyACM0 115200`.
`minicom -D /dev/ttyACM0`
[image]
`screen /dev/ttyACM0 115200`
[image]

jimmy-huang, 2017-08-15:
It might be a bug of miniom that it doesn't support these special characters, I don't see how we could've introduced this bug in our side.