intel/zephyr.js #1329: [aio] aioPin.read and aioPin.readAsync cost more time than before.

qiaojingx, 2017-07-04:
### Description

 aioPin.read and aioPin.readAsync will take more time to finish reading. The efficiency is lower than before

### Test Code

[test-aio.js](https://github.com/01org/zephyr.js/blob/master/tests/test-aio.js);

or simple code:
```
console.log("Wire IO4 to A0!"); 

var aio = require("aio");
var gpio = require("gpio");
var pins = require("arduino101_pins");
var pinA = aio.open({pin: pins.A0});
var pinB = gpio.open({pin: "IO4", mode: 'out'});
var readFlag = 0;

// test AIOPin read

setInterval(function () {
    pinB.write(readFlag);
    console.log("pinB.write:  " + readFlag);
    console.log("pinA.read:  " + pinA.read());
    pinA.readAsync(function (rawValue) {
		console.log("readAsync: " + rawValue);
		console.log("------------------------");
      });

	readFlag = 1 - readFlag;
}, 5000);
```
### Steps to Reproduction

 
### Actual Result

Test result for test-aio.js:
[image]

Test result for simple code: aioPin.read and aioPin.readAsync can't finish reading in time.
[image]
 
### Expected Result

Test result for test-aio.js:
[image]

Test result for simple code: 
[image]

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 418c871 | Arduino 101 | Jul 4, 2017 | Fail
master | a09bd64 | Arduino 101 | Jul 4, 2017 | Pass
 
### Additional Information

jimmy-huang, 2017-07-06:
The issue is that now AIO.read() and readAsync() reads the cached value to speed up the read calls, and doesn't see the updated value until later when it actually changed, we can fix this by reverting the patch, but it will bring back the high latency of ADC reads, so maybe the question for the 0.4 release is which is more important?  Maybe @grgustaf can comment?

qiaojingx, 2017-08-03:
Verified with commit 7054087 + #1404 . This issue is not reproduced.
I will close it after patch merge.

closed 2017-08-07 by grgustaf

qiaojingx, 2017-08-09:
Verified with commit ca7af31 on Arduino101. This issue is not reproduced.

PR #1404 (2017-08-07): [aio] Revert previous AIO read patch
The patch of spawning another thread to read cached
values is not needed as it has a regression issue and
there's a better way to solve the slow read of ADC.

Fixes #1329
commit 9f75ded5 (2017-08-07): [aio] Revert previous AIO read patch (#1404)

The patch of spawning another thread to read cached
values is not needed as it has a regression issue and
there's a bug in the ADC driver that if we set the
sampling delay to 12, it will be underflowing and
have a huge delay, setting to 26 will fix the issue.

Fixes #1329