intel/zephyr.js #1335: Failed to build samples

qiuzhong, 2017-07-07:
## Description
After I update the zephyr.js to [0334065](https://github.com/01org/zephyr.js/commit/033406567c9c00fe039675547767134daab9200b), I can't build the samples any more.

## Steps to reproduce
1. `git clone https://github.com/01org/zephyr.js`
2. `cd zephyr.js && source zjs-env.sh && make update`
3. `make ROM=256 JS=samples/OcfServer.js`

## Expected Result
No error for the sample build.

## Actual Result
```
/bin/sh: 1: scripts/gen_idt/gen_idt: not found
/home/qiuzhong/Workspace/Work/hj/zephyr.js/deps/zephyr/arch/x86/Makefile.idt:35: recipe for target 'staticIdt.o' failed
make[3]: *** [staticIdt.o] Error 127
make[3]: Leaving directory '/home/qiuzhong/Workspace/Work/hj/zephyr.js/outdir/arduino_101'
Makefile:178: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
make[2]: Leaving directory '/home/qiuzhong/Workspace/Work/hj/zephyr.js/deps/zephyr'
/home/qiuzhong/Workspace/Work/hj/zephyr.js/deps/zephyr/Makefile.inc:82: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/qiuzhong/Workspace/Work/hj/zephyr.js'
Makefile:196: recipe for target 'zephyr' failed
make: *** [zephyr] Error 2

```

## Build Environment:
* Zephyr SDK : 0.9.1
* Zephyr.js commit:  [0334065](https://github.com/01org/zephyr.js/commit/033406567c9c00fe039675547767134daab9200b)

qiuzhong, 2017-07-07:
We found this error is caused by line 35 of `zephyr.js/deps/zephyr/arch/x86/Makefile.idt`:

`        GENIDT := scripts/gen_idt/gen_idt`

If we add a `$(ZEPHYR_BASE)` prefix for the GENIDT variable, then there is no such error.

wanghongjuan, 2017-07-07:
If directly clone the master branch(`0334065`), this issue won't be reproduced.
```
git clone https://github.com/01org/zephyr.js.git
cd zephyr.js; source zjs-env.sh;make update; source deps/zephyr/zephyr-env.sh
make ROM=256 JS=sample/OcfServer.js
```

jimmy-huang, 2017-07-07:
Alternatively:
rm -rf deps/zephyr
make update

jimmy-huang, 2017-07-11:
I am closing this as it should be resolved.

closed 2017-07-11 by jimmy-huang