intel/zephyr.js #1345: [uart] only write the first four letters after update Zephyr 1.8.0

haoxli, 2017-07-11:
# Description
After update Zephyr to 1.8.0, there are only first 4 letters can output in console window when write a buffer. 

# Test Code
[UART.js](https://github.com/01org/zephyr.js/blob/master/samples/UART.js)
```
uart.write(new Buffer('UART write succeeded, echoing input.\r\n'));
```

# Steps to Reproduction
1. `$ make JS=samples/UART.js `
2. `$ make dfu`
3. Reset the board
4. In another Terminal: `minicom -D /dev/ttyACM0` 

# Expected Result
[image]

# Actual Result
[image]
Enter a string "abc", output is only the first letters of "Data recv: abc"
[image]

# Test Builds
Branch | Build ID | Target Device| Date | Result
---|---|---|---|---
master | 0334065| Arduino 101| July 11, 2017 | Fail
master | 4e44cc6| Arduino 101| July 11, 2017 | Pass

brianjjones, 2017-07-18:
Apparently the A101 would occationally repeat bytes if more than 4 bytes were sent at a time using CDC_ACM, so Zephyr made it so you could only send 4 bytes at once (ZEP-2074).  I filed a bug (jira.zephyrproject.org/browse/ZEP-2401) to track this.  In the meantime I created a fix for UART that causes it to keep sending 4 bytes at a time until the entire buffer is sent. See https://github.com/01org/zephyr.js/pull/1377

qiaojingx, 2017-07-19:
Verified with commit af7eadd + #1377. This issue is not reproducible.

commit 6241a721 (2017-07-24): [uart] Fix for issue #1345, only 4 bytes sent to UART (#1377)

[uart] Workaround for sending more than 4 bytes sent to UART

Fixes #1345
closed 2017-07-24 by grgustaf