intel/zephyr.js #1348: [ws] Websocket connection cannot setup on A101

qiaojingx, 2017-07-11:
### Description

Websocket connection cannot setup on A101. BLE connection will break after waiting websocket connection for a while. 
 
### Test Code

[WebSocketServer.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer.js); [test-ws-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ws-server.js); 

### Steps to Reproduction

1. `make JS=samples/websockets/WebSocketServer.js ROM=256`
2. `make dfu`
3. connect via BLE then add the IP route for the Arduino's 6lowpan address
    `ip -6 route add 2001:db8::/64 dev bt0`
4. execute `node samples/websockets/NodeWebsocketClient.js` in terminal

### Actual Result

Connection cannot setup. After a while, ble connection will break.
[image]

### Expected Result

Websocket connection setup successfully.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | c4c6c01 | Arduino 101 | Jul 11, 2017 | Fail

### Additional Information

cuiyanx, 2017-07-12:
I follow qiaojing steps to reproduce this issue, I got another issue after connected and waited for a few minutes.
```
Connected
***** Unhandled interrupt vector *****
Current thread ID = 0xa800c300
Faulting segment:address = 0x0008:0x40000029
eax: 0xa800c3a8, ebx: 0xa800c3a9, ecx: 0x00000000, edx: 0x00000003
esi: 0xa800c3e0, edi: 0xa800c3fc, ebp: 0xa800f7f4, esp: 0xa800f7db
eflags: 0x246
Fatal fault in thread 0xa800c300! Aborting.
CONNECTION

```

brianjjones, 2017-07-20:
A fix can be found here https://github.com/01org/zephyr.js/pull/1384

qiaojingx, 2017-07-21:
Verified with commit 5c65f11, this issue is not reproducible. New issue(#1386) is reported for throwing error `(ERROR) cannot acquire send_pkt`.
Closed this one.

closed 2017-07-21 by qiaojingx