intel/zephyr.js #1358: [DHCP] fail to build samples/DHCP.js for K64F

qiaojingx, 2017-07-13:
### Description

` undefined reference to 'zjs_net_config_init'` error occurs when building [DHCP.js](https://github.com/01org/zephyr.js/blob/master/samples/DHCP.js) sample for FRDM-K64F. 

### Test Code

[DHCP.js](https://github.com/01org/zephyr.js/blob/master/samples/DHCP.js)
 
### Steps to Reproduction

 1. `make BOARD=frdm_k64f JS=samples/DHCP.js`

### Actual Result

 
[image]

### Expected Result

 Build DHCP.js sample successfully.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 1ff882b | FRDM-K64F | Jul 13, 2017 | Fail
 
### Additional Information

qiaojingx, 2017-07-21:
Update issue description since `net-config` module should also be supported on Arduino 101.

jprestwo, 2017-07-21:
This sample was carry over from a stand alone DHCP module. The net-config module must be used in conjunction with another networking module (net/ws/dgram). These other modules include the needed config options for networking in general, which DHCP needs to work. Because of this, DHCP.js will not work, I pushed #1389 to remove it. To test DHCP you can use TCPEchoServ4_DHCP.js.

The next issue is the support between boards/targets. Currently only the FRDM board supports DHCP i.e. the `dhcp()` API is only defined on that target. This was done because it is currently the only "real" use case for it. The Arduino 101 uses 6lowpan which doesn't seem like DHCP would work (maybe I am wrong). @pfalcon did say you could do DHCP on QEMU, though I don't think that is a very good use case either, but perhaps for testing. It would be trivial to add DHCP support on QEMU if thats desired.

closed 2017-07-24 by grgustaf

qiaojingx, 2017-07-25:
Verified with commit 5726f4e. DHCP.js sample has been removed.

PR #1389 (2017-07-24): [samples] Remove DHCP.js sample
DHCP can only be used with another networking module. This sample
does not include net/ws/dgram therefore it does not work. DHCP
should be tested using one of the networking module samples, like
TCPEchoServ4_DHCP.js

Fixes #1358
commit 858f562e (2017-07-24): [samples] Remove DHCP.js sample (#1389)

DHCP can only be used with another networking module. This sample
does not include net/ws/dgram therefore it does not work. DHCP
should be tested using one of the networking module samples, like
TCPEchoServ4_DHCP.js

Fixes #1358