intel/zephyr.js #1363: GPIO memory leak

brianjjones, 2017-07-13:
The pin object created at line 301 isn't being freed when the javascript stops.  This is likely due to another reference of pin_object being created somewhere, so jerryscript never calls zjs_gpio_free_cb.

grgustaf, 2017-08-15:
I compiled ashell with TRACE=on and zjs_print_mem_stats() in the 'stop' function and I don't see a leak when running tests/test-gpio.js. After the JS runs, I see two GPIO pins left allocated; but this is to be expected because they're in the global scope. If I type 'stop' so that the global scope is freed up, and it hits my extra mem stats print, the objects are gone. And running it repeatedly I don't see the leak growing.

(I removed some modules from zjs_ashell.json to make sure it would fit.)

brianjjones, 2017-08-29:
I still see them myself if I wrap the entire test in a function, making those GPIO pins local.

index 0 /home/bjones/ZJS/master1.8/src/zjs_callbacks.c - zjs_init_callbacks:165: 0xa800bf98
index 1 /home/bjones/ZJS/master1.8/src/zjs_callbacks.c - zjs_add_c_callback:363: 0xa800bfe0
index 3 /home/bjones/ZJS/master1.8/src/zjs_callbacks.c - add_callback_priv:209: 0xa800c050
index 4 /home/bjones/ZJS/master1.8/src/zjs_gpio.c - zjs_gpio_open:304: 0xa800c070
index 6 /home/bjones/ZJS/master1.8/src/zjs_callbacks.c - zjs_add_c_callback:363: 0xa800c0d0
index 7 /home/bjones/ZJS/master1.8/src/zjs_gpio.c - zjs_gpio_open:304: 0xa800c0f0
index 8 /home/bjones/ZJS/master1.8/src/zjs_callbacks.c - zjs_add_c_callback:363: 0xa800c120
zjs_free_callback:244: freeing 0xa800c0d0
zjs_free_callback:244: freeing 0xa800c050

grgustaf, 2017-09-07:
Postponing since it's not a P1.

grgustaf, 2018-01-22:
I thought this was trivial fix but I get crashes when I use the above patch, which haven't been easy to track down. It seems like I might be getting a double free.

PR #1764 (2018-02-01): [gpio] Fix memory leak through unnecessary acquire
Since the native pointer struct held a reference to the pin object,
neither would ever get freed.

Fixes #1363
commit f158821f (2018-02-01): [gpio] Fix memory leak through unnecessary acquire

Since the native pointer struct held a reference to the pin object,
neither would ever get freed.

Fixes #1363
closed 2018-02-01 by jimmy-huang