intel/zephyr.js #1367: [PnP] All of ARC ROM samples increase obviously compared with master commit 05fda87.

hanyia, 2017-07-17:
Test with latest master commit on zephyr board, found all ARC ROM samples has more than ~5% gap compared with the last commit. SDK version is v0.9.1. Detail info as below table:

Device | KPI | ww29 master: af7eadd | ww28 master: 05fda87 | ww29 vs ww28 |
----|------|----|------|----
Arduino101 | Zephyr_Helloworld_ARC_ROM(Byte)  | 6156 | 5720 | 7.62%|
Arduino101 | Zephyr_I2C_ARC_ROM(Byte)   | 20688 | 16108 | 28.43% |
Arduino101 | Zephyr_WebBluetoothGroveLcdDemo_ARC_ROM(Byte)  | 26928 | 21296 | 26.45% |

jimmy-huang, 2017-07-18:
This is a know issue, which we have to deal with a upstream Zephyr (ZEP-2363) with 1.8 and 0.9.1 SDK, and we will have to build ARC with -O2 instead of -Os.  This won't be fixed for 0.4 release, we'll have to wait till Zephyr upstream fixes this or a new SDK.

jimmy-huang, 2017-08-14:
We should follow up with Zephyr on fixing this bug.

jimmy-huang, 2018-01-10:
This is already fixed upstream, so I am closing this and re-open if needed.

hanyia, 2018-01-24:
Device | KPI |ww04 master:e720399 | ww29 master: af7eadd | ww28 master: 05fda87 | ww20 master:e24821c | ww19 master:3950beb
-- | -- | -- | -- | -- | -- | -- 
Arduino101 | Zephyr_Helloworld_ARC_ROM(Byte) | 6436 | 6156 | 5720 | 10496 | 12684
Arduino101 | Zephyr_Helloworld_ARC_RAM(Byte) | 3920 | 4940 | 4940 | 6712 | 5592
Arduino101 | Zephyr_Helloworld_ROM(Byte) | 100756 | 97744 | 97744 | 100465 | 102268
Arduino101 | Zephyr_Helloworld_RAM(Byte) | 26884 | 24704 | 24704 | 26508 | 26508
Arduino101 | Zephyr_I2C_ARC_ROM(Byte) | 20216 | 20688 | 16108 | 20072 | 16464
Arduino101 | Zephyr_I2C_ARC_RAM(Byte) | 4604 | 5624 | 5624 | 6848 | 5728
Arduino101 | Zephyr_I2C_ROM(Byte) | 112728 | 108653 | 108653 | 109778 | 111452
Arduino101 | Zephyr_I2C_RAM(Byte) | 28872 | 26752 | 26572 | 26572 | 26572
Arduino101 | Zephyr_WebBluetoothGroveLcdDemo_ARC_ROM(Byte) | 24820 | 26924 | 21296 | 24436 | 20220
Arduino101 | Zephyr_WebBluetoothGroveLcdDemo_ARC_RAM(Byte) | 5004 | 7184 | 7184 | 7288 | 6168
Arduino101 | Zephyr_WebBluetoothGroveLcdDemo_ROM(Byte) | 174388 | 171813 | 171845 | 170499 | 172184
Arduino101 | Zephyr_WebBluetoothGroveLcdDemo_RAM(Byte) | 38900 | 36288 | 36288 | 35024 | 34916
SDK |  | 0.9.2 | 0.9.1 | 0.9.1 | 0.9 | 0.8.2

hanyia, 2018-01-24:
@jimmy-huang I attached complete date as below, it seems that update SDK to 9.2, not fix this issue, so report it.

jimmy-huang, 2018-01-24:
@hanyia Ok, looks like they didn't fix it upstream, I just submitted a pr https://github.com/zephyrproject-rtos/zephyr/pull/5815, so this will be fixed in the next release.

PR #1789 (2018-01-25): [arc] Set build optimization flag to -Os
This will be a work-around to manually set the build
flag from -Os to -O2 until patch is merged into Zephyr
upstream.

Fixes #1367
commit 52fa54ab (2018-01-25): [arc] Set build optimization flag to -Os (#1789)

This will be a work-around to manually set the build
flag from -Os to -O2 until patch is merged into Zephyr
upstream.

Fixes #1367
closed 2018-01-25 by grgustaf