intel/zephyr.js #1368: [ashell] acm> does not print at the end of console log

qiaojingx, 2017-07-17:
### Description

`acm>` does not print at the end of console log. It affect the next input.

### Test Code

[ashell](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md).
 
### Steps to Reproduction

1. `make ashell ROM=256`
2. `make dfu`
3.  Then use the following command to connect to the ashell from a terminal:
     `$ screen /dev/ttyACM0 115200`
 
### Actual Result

acm> does not print at the end of console log.
[image]
 
### Expected Result

`acm>` should print as end 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | af7eadd| Arduino 101 | Jul 17, 2017 | Fail
 
### Additional Information

brianjjones, 2017-08-05:
The fix for this can be found here
https://github.com/01org/zephyr.js/pull/1431

qiaojingx, 2017-08-09:
Verified with commit ca7af31 + #1431. This issue is not reproduced.

qiaojingx, 2017-08-24:
Verified with commit 528a6f0. This issue is fixed. Close

closed 2017-08-24 by qiaojingx