intel/zephyr.js #1369: [ws] Can not trigger connection event before ws client send message

cuiyanx, 2017-07-18:
### Description

 Before ws client send some message, even if the ws connection is successful, the connection event will not be triggered.

### Test Code

[tests/test-ws-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ws-server.js)
[tests/test-ws-client-node.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ws-client-node.js)

 

### Steps to Reproduction

1. Run [tests/test-ws-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ws-server.js) on a101 or K64f.
2. Run [tests/test-ws-client-node.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ws-client-node.js) on linux.
3. Check log and the program will be hang.
4. Add `ws.send("hello world");` on line 8 in  [tests/test-ws-client-node.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ws-client-node.js).
5. Will see the program work well.
 

### Actual Result

Will be hang.
 

### Expected Result

Work well and can trigger connection event correctly.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | af7eadd64 | Arduino 101 | July 18, 2017 | Fail
master | af7eadd64 | K64f | July 18, 2017 | Fail

 

### Additional Information

jprestwo, 2017-07-24:
This exposes an issue with events that @grgustaf is currently working on. There are two main issues:

1. Triggering an event before one has been created. This is what is happening here. The `connection` event has not yet triggered, so the `message` event has not been registered. Then data comes in and cannot successfully trigger `message`.

2. Triggering the same event twice before its been processed. Not an issue with this bug but still an issue. Here, a event_trigger_t struct is being allocated and set as the event callback handle. This will get freed in `post_event`, but before the event is processed it is triggered again, overwriting the handle. This causes a memory leak.

Both issues will be addressed by @grgustaf.

jimmy-huang, 2017-08-02:
This should be fixed with #1424, can you verify?

qiaojingx, 2017-08-03:
Verified with commit 4d9b084. This issue is not reproduced.
Platform: Arduino101, FRDM-K64F, QEMU.

cuiyanx, 2017-08-03:
fixed and closed

commit 776330fb (2017-08-07): [general] Increase call stack size to improve BLE stability (#1424)

The callstack of 2k is causing segfault in BLE stack,
for example, when it starts physical web advertising,
or the Websocket over 6lowpan, increasing
it to 4k fixes the segfault.

Fixes #1369 #1421
closed 2017-08-03 by cuiyanx