intel/zephyr.js #1371: [net] net.on('up') function is broken

qiaojingx, 2017-07-18:
### Description

Got `Error: Expected a function` error when test with function  
```
net.on("up", function() {
});
```
### Test Code

[TCPClient6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPClient6.js);
[test-tcp6-client.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp6-client.js)
 
### Steps to Reproduction

1. add console.log to [TCPClient6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPClient6.js); as below:
[image]
2. `make JS=samples/TCPClient6.js ROM=256`
3. `make dfu`
 
### Actual Result

Got `Error: Expected a function` error 
[image]

 ### Expected Result

No error thrown. 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | af7eadd | Arduino 101 | Jul 18, 2017 | Fail
 
### Additional Information

jprestwo, 2017-07-20:
The networking up/down events were moved (and renamed to netup/netdown) into the net-config module. These scripts must not have been updated. You will have to add the following code:
```
var net_cfg = require('net-config');
net_cfg.on('netup', function() {
    // rest of code here
}
```

qiaojingx, 2017-07-21:
@jprestwo ， Ok then，please update [docs/net.md](https://github.com/01org/zephyr.js/blob/master/docs/net.md). I will close this issue after document updated.

commit 575163a7 (2017-07-24): [docs] Removed onup event from net, now in net-config (#1388)

Fixes #1371
closed 2017-07-24 by grgustaf