intel/zephyr.js #1373: [dhcp] NetConfig.dhcp is broken in qemu

qiaojingx, 2017-07-18:
### Description

When test with functions `net_cfg.dhcp(function(address, subnet, gateway) { });` in qemu, `Error: Expected a function error` will be thrown. 

### Test Code

[TCPEchoServ4_DHCP.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ4_DHCP.js)
[WebSocketServer4.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer4.js)
 
### Steps to Reproduction

1. add console.log to [TCPEchoServ4_DHCP.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ4_DHCP.js) as below:
[image]
2. Start net-tools by following [Networking with QEMU](https://github.com/01org/zephyr.js#networking-with-qemu)
3. `make BOARD=qemu_x86 qemu JS=samples/TCPEchoServ4_DHCP.js`

### Actual Result

Got `Error: Expected a function error` on QEMU:
[image]

 ### Expected Result

No error thrown. 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | af7eadd | QEMU | Jul 18, 2017 | Fail
 
### Additional Information

jprestwo, 2017-07-18:
DHCP is not enabled (and probably not possible) with QEMU as it expects a static IP of 192.0.2.0.

pfalcon, 2017-07-18:
> DHCP is not enabled (and probably not possible) with QEMU

Well, it's possible, you just need to run dhcp server on your host, which isn't done by default and rarely done at all (a typical workstation is a dhcp client, server is some router/access point). Zephyr net-tools should contain some configs/instructions to run it though.

qiaojingx, 2017-08-25:
Does DHCP plan to be enabled with QEMU before 0.4 release? It blocks dhcp related tests on Linux(Actually run in QEMU).

grgustaf, 2017-09-15:
This is not going to be a high priority for the dev team, given that it works on real hardware. If it was broken on real hardware we might have a stronger incentive to try to get it working with QEMU. So I'm going to lower the priority further, and I wouldn't plan QA around this unless you want to look into enabling it yourselves.

I did *enable* DHCPv4 for all builds w/ commit e8079103, so the code should build; whether it works is maybe just a configuration issue now.

grgustaf, 2017-09-15:
The bug as stated is really fixed now though, you should no longer see "Expected a function".

qiaojingx, 2017-09-20:
Verified with commit 174f35f. This issue is fixed.
DHCP works in qemu.
[image]
Close.

closed 2017-09-20 by qiaojingx