intel/zephyr.js #1383: [ocf] Failed to get multiple resources which use same resourceType

wanghongjuan, 2017-07-20:
## Description
API describe function [`findResources(ClientOptions options, optional FoundListener listener);`](https://github.com/01org/zephyr.js/blob/master/docs/ocf.md#clientfindresources): `Find remote resources matching options filter`, but only got the last register temp resource, light is undefined. 

## Test Code
Base on the test code of #1197, updated all resourceTypes to `resourceTypes:["core.res"]`

## Steps to Reproduction
1. Build [/samples/OcfMultiSensor.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfMultiSensor.js) on a101
2. Connect with BLE
3. Build and run simple test on linux
4. Check console log

## Actual Result
Only got the last register temp resource, light is undefined.

## Expected Result
Can find the both resources.

## Test Builds
| Branch | Commit id | Target Device | Test Date | Result |
| :---:     | :---: |    :---:    | :---: | :---: |
| master | de91ade | Arduino 101 | July 19, 2017 | Fail |

cuiyanx, 2017-07-20:
Simple test as ocf client on linux:
```
var ocf = require('ocf');
var client = ocf.client;

ocf.start();

client.findResources({ resourceType:"core.res" }).then(function(resource) {
    console.log("findResources 'core.res' was successful, deviceId=" + resource.deviceId);

    client.retrieve(resource.deviceId, {observable: false}).then(function(res) {
        console.log("ResourcePath:" + res.resourcePath);
        console.log("light: " + res.properties.light);
        console.log("ResourcePath:" + res.resourcePath);
        console.log("temp: " + res.properties.temp);
    }, function(error) {
    });
}, function(error) {
    console.log("findResources() returned an error: " + error.name);
});
```

Log:
```
ResourcePath:/a/temp
light: undefined
ResourcePath:/a/temp
temp: 4091
```
Maybe more clear.

wanghongjuan, 2017-07-20:
I think maybe we can enhance this method by updating the options parameter to optional, if there is no parameter input, the function `findResources` can find all resources, since maybe there are multiple resources on server file with different resourceType/resourcePath(such as the sample/OcfMuiltServer.js),. Meanwhile, we can avoid test code redundancy, just like the test code of #1197 for client.js, must to invoke two times `findResources`.

cuiyanx, 2017-07-20:
but we can add `the options parameter` to find resources.
Simple test as client on linux:
```
var ocf = require('ocf');
var client = ocf.client;

ocf.start();

client.findResources({ resourceType:"core.res", resourcePath:"/a/light" }).then(function(resource) {
    console.log("findResources 'core.light' was successful, deviceId=" + resource.deviceId);

    client.retrieve(resource.deviceId, {observable: false}).then(function(res) {
        console.log("ResourcePath:" + res.resourcePath);
        console.log("light: " + res.properties.light);
    }, function(error) {
    });
}, function(error) {
    console.log("findResources() returned an error: " + error.name);
});

setTimeout(function() {
    client.findResources({ resourceType:"core.res", resourcePath:"/a/temp" }).then(function(resource) {
        console.log("findResources 'core.temp' was successful, deviceId=" + resource.deviceId);

        client.retrieve(resource.deviceId, {observable: false}).then(function(res) {
            console.log("ResourcePath:" + res.resourcePath);
            console.log("temp: " + res.properties.temp);
        }, function(error) {
        });
    }, function(error) {
        console.log("findResources() returned an error: " + error.name);
    });
}, 3000);

Log:
```
findResources 'core.light' was successful, deviceId=08547a09-5f54-4a09-6d54-7a09fb547a09
ResourcePath:/a/light
light: 4095
findResources 'core.temp' was successful, deviceId=08547a09-5f54-4a09-6d54-7a09fb547a09
ResourcePath:/a/temp
temp: 4091
```

```

wanghongjuan, 2017-07-20:
@cuiyanx I understand what's your mean, I just suggest we can optimize the test code, update the required `option parameter` to optional is an optional optimization solution, you can choose add it or not, this is not an issue.

jimmy-huang, 2017-07-20:
Ok, I think there's a design issue of the api, the promise when resolved only contains 1 resource, so it will never give you back more than 1 resource.  The correct way to find multiple resources, is that you pass in the listener function when calling findResources(), and the promise should resolve with no arguments, like:

function onResourcefound(resource) {
    console.log("resource found, deviceId=" + resource.deviceId);

    client.retrieve(resource.deviceId, {observable: false}).then(function(res) {
        console.log("light: " + res.properties.light);
    }, function(error) {
    });
}

client.findResources({ resourceType:"core.res" }, onResourcefound).then(function() {
    console.log("findResources was successful);
}, function(error) {
    console.log("findResources() returned an error: " + error.name);
})

This means we'll need to update the client API to reflect this, so @zolkis, maybe you can give feedback here.

jimmy-huang, 2018-01-23:
This requires API change so won't be fixed for 0.5 release.