intel/zephyr.js #1385: [ws] If disconnect ws connection, you will never create a new connection

cuiyanx, 2017-07-21:
### Description

I disconnect ws connection and will re-connect, but ws server and client look like be hang. So I close ws client and get an error.
 

### Test Code

[/samples/websockets/WebSocketServer.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer.js)
[/samples/websockets/NodeWebSocketClient.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/NodeWebSocketClient.js)

 

### Steps to Reproduction

1. Run WebSocketServer.js on a101
2. Run NodeWebSocketClient.js on linux
3. Disconnect this ws connection
4. Re-connect ws connection
5. Close ws client
6. Check log info.
 

### Actual Result

Error info:
```
 src/zjs_event.c:358 zjs_trigger_event():
(ERROR) callback_id not found
```

### Expected Result

 Work well and can be re-connected.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 5c65f1102 | Arduino 101 | July 21, 2017 | Fail

 

### Additional Information

qiaojingx, 2017-08-03:
Verified with commit 4d9b084. Still can't create a new connection after disconnect ws connection.
But no error is print now.
[image]

brianjjones, 2017-08-04:
This seems to be due to websockets not handling close properly.  Its not emitting the close signal or handling it.  I'm working on a fix now.

brianjjones, 2017-08-04:
Unfortunately I still can't seem to get this working.  I altered the NodeWebSocketClient.js to  call ws.terminate() after about 10 seconds, and then create a new WebSocket connection after another 5.  It always hangs at the same spot during the second connect, zjs_web_sockets.c:220.  I suspect it may be a Zephyr bug.  Whoever looks at this next might see if there are any zephyr samples that could demonstrate being able to reconnect or have multiple clients.

grgustaf, 2017-08-10:
Looking into this bug, I found that the sample doesn't work at all on A101+Ethernet. I'm pretty sure it was working when I checked in the big event patch last week though. After lots of debugging with @jprestwo's help, it appears that adding NET_BUF_DATA_SIZE=512 like @jimmy-huang did for OCF recently fixes that. We also discovered a June 16th patch in Zephyr, right after 1.8 release, that had to do with net_pkt fragmentation, but ultimately that didn't seem to be needed.

There seems to be a bug in the net_pkt fragmentation that this exposes. Oddly it doesn't seem to happen over Bluetooth, even though it's Bluetooth that Jimmy first had the problem with in OCF code.

Something's a bit flaky; occasionally the sample won't work, and occasionally it stops at CONNECTION and doesn't get farther. But w/ just this 512 byte patch I just now ran the node code about 20 times w/o rebooting the device and it kept working. I'll get back to looking at how it works over BT tomorrow.

qiaojingx, 2017-08-11:
Verified with commit 247b51a. Websocket is able to reconnect on Arduino101 over BT.
[image]
Waiting for ENC28J60 Ethernet Board to verify websocket samples on Arduino101+ Ethernet.

closed 2017-08-14 by jimmy-huang

qiaojingx, 2017-08-16:
Verified with commit 212747a. This issue is not reproduced on Arduino101 over BT

cuiyanx, 2018-01-10:
Fixed and closed

PR #1460 (2017-08-14): [websockets] Increase net buf data size to work around issues
There appears to be a bug in Zephyr's net buf fragmentation code,
or else this shouldn't make a difference. Need to look into that
further.

Also, clean up some code and output in the node sample code.

Fixes an unreported issue where the websocket sample doesn't work at
all on A101 w/ Ethernet currently.

Fixes #1385
commit 24395258 (2017-08-14): [websockets] Increase net buf data size to work around issues (#1460)

There appears to be a bug in Zephyr's net buf fragmentation code,
or else this shouldn't make a difference. Need to look into that
further.

Also, clean up some code and output in the node sample code.

Fixes an unreported issue where the websocket sample doesn't work at
all on A101 w/ Ethernet currently.

Fixes #1385