intel/zephyr.js #1393: Key 'JSON' is not in deps list

qiaojingx, 2017-07-24:
### Description

This is a regression issue. Building [PMESaveRestore.js](https://github.com/01org/zephyr.js/blob/master/samples/PMESaveRestore.js) and [test-pme-save-restore.js](https://github.com/01org/zephyr.js/blob/master/tests/test-pme-save-restore.js) failed for a101.

### Test Code

[PMESaveRestore.js](https://github.com/01org/zephyr.js/blob/master/samples/PMESaveRestore.js);
[test-pme-save-restore.js](https://github.com/01org/zephyr.js/blob/master/tests/test-pme-save-restore.js)
 
### Steps to Reproduction

 1. `make JS=samples/PMESaveRestore.js`

### Actual Result

[image]
 
### Expected Result

 Building succeed.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 5c65f11 | Arduino 101 | Jul 24, 2017 | Fail
master | af7eadd | Arduino 101 | Jul 18, 2017 | Pass
 
### Additional Information

jimmy-huang, 2017-07-24:
Did something changed recently on analyze?  I think this should be JSON not json as it's checking for the use of JSON object or constructors in JS, usually a call to JSON.parse()?

closed 2017-07-24 by jimmy-huang

qiaojingx, 2017-07-25:
Verified with commit 5726f4e. This issue is fixed.

PR #1395 (2017-07-24): [analzye] Ensure dlist is composed only of module names
Fixes #1393
commit 5726f4ec (2017-07-24): [analzye] Ensure dlist is composed only of module names (#1395)

Fixes #1393